justcallmekoko/ESP32Marauder #421: Satellite Counter

wh1ter0z, 2023-12-09:
**Describe the bug**
Satellite counter when came down from 10 to 9 it will be 90 not 9 

**To Reproduce**
Steps to reproduce the behavior:
1. Status bar left side 

**Screenshots**
[image]

**Marauder (please complete the following information if applicable):**
 - Firmware version: last version
 - Hardware version: V4 OG

**Additional context**
Just add an space after numbers to be fix

alinder, 2024-01-05:
I'm seeing this issue as well on the Marauder v6.1 running 13.6.  

It's easy to reproduce, just go outside in an open area, wait for the satellite count to hit 10, and then remove the antenna.  As the GPS loses satellite signals, the count will drop to 90, 80, 70, 60, etc.

wh1ter0z, 2024-01-05:
> I'm seeing this issue as well on the Marauder v6.1 running 13.6.  
> 
> It's easy to reproduce, just go outside in an open area, wait for the satellite count to hit 10, and then remove the antenna.  As the GPS loses satellite signals, the count will drop to 90, 80, 70, 60, etc.

Yes 

Its on menu function i fixed

wh1ter0z, 2024-01-05:
> I'm seeing this issue as well on the Marauder v6.1 running 13.6.  
> 
> It's easy to reproduce, just go outside in an open area, wait for the satellite count to hit 10, and then remove the antenna.  As the GPS loses satellite signals, the count will drop to 90, 80, 70, 60, etc.

https://github.com/justcallmekoko/ESP32Marauder/blob/master/esp32_marauder%2FGpsInterface.cpp#L499

Search this function and add 2x space :

.... getNumSatsString() + "  " ...