justcallmekoko/ESP32Marauder #425: M5stickc plus crashing due SD Card

ThiagoMathiaz, 2023-12-12:
**Describe the bug**
When trying to open options like evil portal, m5stickc plus restarts.
I'm using version 13.04.
When I use version 13.05, m5stickc plus restarts at boot time.

**To Reproduce**
Steps to reproduce the behavior:
1. See the video

https://github.com/justcallmekoko/ESP32Marauder/assets/62333921/b306f4bb-e28d-4be6-a82b-a6ef69a6678a

**Marauder (please complete the following information if applicable):**
 - Firmware version: v13.04
 - Hardware version: M5stickc plus

**Additional context**
Add any other context about the problem here.

pitec0, 2023-12-13:
Same here, but my loading its in loop and dont start the marauder.
[image]

Masterbudz01, 2023-12-23:
me too please help

CavaleiroCinza, 2023-12-30:
Guys, the problem is the marauder versions. Try installing older versions 13.3. New versions like 13.5/13.6 are not working. I hope I helped, thanks!

CavaleiroCinza, 2023-12-30:
> **Describe the bug**
> When trying to open options like evil portal, m5stickc plus restarts.
> I'm using version 13.04.
> When I use version 13.05, m5stickc plus restarts at boot time.
>
> **To Reproduce** [...]

Fala Matuê de Chernobyl, então...
O problema é que as versões não foram bem otimizadas para o m5stick c plus, nem as novas versões como a 13.5/13.6 funcionam (elas ficam reiniciando o m5stick quando identificam o cartão sd, caso você remova o módulo vai voltar a funcionar normalmente) Então a solução é usar versões antigas ou seja, da 13.4 para baixo.
O evil portal não funciona porquê falta um código aonde o marauder acesse o cartão sd e possa escrever nele; ai quando o programa não encontra o código, faz com que o m5 reset. 

Agora é aguardar atualizações para que isso seja resolvido ou meter a mão na massa em casa e corrigir por si mesmo.
Tmj!

Masterbudz01, 2023-12-30:
i have tried with 13.3 version and all obsolete apps works, but when i start evil portal, device crash and restarts disconnecting SD card.. i don't know why.

pitec0, 2024-01-05:
> > **Describe the bug**
> > When trying to open options like evil portal, m5stickc plus restarts.
> > I'm using version 13.04.
> > When I use version 13.05, m5stickc plus restarts at boot time.
> > **To Reproduce**
> > Steps to reproduce the behavior:
> > 
> > 1. See the video
> > 
> > 
> >   
> >     
> >     
> > 
> >     IMG_3886.MOV
> >     
> >   
> > 
> >   
> > 
> >   
> > 
> > 
> > **Marauder (please complete the following information if applicable):**
> > 
> > * Firmware version: v13.04
> > * Hardware version: M5stickc plus
> > 
> > **Additional context**
> > Add any other context about the problem here.
> 
> Fala Matuê de Chernobyl, então... O problema é que as versões não foram bem otimizadas para o m5stick c plus, na verdade foi mal portado mesmo. Nem as novas versões como a 13.5/13.6 funcionam (elas ficam reiniciando o m5stick quando identificam o cartão sd, caso você remova o módulo vai voltar a funcionar normalmente) Então a solução é usar versões antigas ou seja, da 13.4 para baixo. O evil portal não funciona porquê falta um código aonde o marauder acesse o cartão sd e possa escrever nele; ai quando o programa não encontra o código, faz com que o m5 reset.
> 
> Agora é aguardar atualizações para que isso seja resolvido ou meter a mão na massa em casa e corrigir por si mesmo. Tmj!

Versão 13.4 não está disponível no m5burner, e a 13.3 não funciona nenhum attack =\

CavaleiroCinza, 2024-01-05:
> Versão 13.4 não está disponível no m5burner, e a 13.3 não funciona nenhum attack =\

Cara, está funcionando sim! Você está com o módulo sd? A maioria dos ataques só funciona com o sd. 
Tanto a 13.3 e a 13.4 estão funcionando perfeitamente!

ThiagoMathiaz, 2024-01-05:
> Cara, está funcionando sim! Você está com o módulo sd? A maioria dos ataques só funciona com o sd. Tanto a 13.3 e a 13.4 estão funcionando perfeitamente!

Nenhuma dessas funcionou, modulo foi reconhecido, mas nao deu nenhum ataque

pitec0, 2024-01-14:
> ma dessas funcionou, modulo foi reconhecido, mas nao deu

Conversei com Marivaldo, ele ja solicitou pull da correção da versão atual ficar reiniciando com o sd, meu sd ta filé, tenho 2 adaptador, os 2 reconhece, mas fica em looping no boot na versão atual ou na versão antiga, não realiza os ataques, agora é esperar. Enquanto isso da pra "brincar" com outros coisas no m5 :D