justcallmekoko/ESP32Marauder #434: M5Stickc Plus2 and Marauder?

kazz2020, 2023-12-16:
Hi Does someone tried M5Stickc PLUS2 with Marauder?

filac001, 2023-12-19:
idk u try it

Masterbudz01, 2023-12-23:
yes but i have an sd card issue and i cant use evil portal

kazz2020, 2023-12-23:
What excactly issue?

Masterbudz01, 2023-12-25:
when i have connected sd correctly with arduino cables, marauder firmware crash and reboots in loop. i have tried old firmware, SD finded but when i start any app marauder crash instantly and reboot.

Masterbudz01, 2023-12-25:
same issue here: https://github.com/justcallmekoko/ESP32Marauder/issues/425

if u have any info please help

sunsu123123, 2023-12-26:
It doesn't seem to work. I put the firmware into the m5Stickc Plus2, but it continues to not boot and stays on a black screen during the reset process.

kazz2020, 2023-12-27:
[image]

Look at TFT pin difference. I think we need compile fw with this pin change. I have problems compile marauder fw at all so maybe someone will try?

devpedromgoncalves, 2023-12-28:
Hi. I'm facing the same issue as @sunsu123123. Does anyone installed and boot marauder successfully on M5StickC Plus 2?
I can install it but I can't reboot and the screen stays black for ever... With other firmwares it doesn't happen, so should we wait for a marauder update on this? :/

kazz2020, 2023-12-28:
> Hi. I'm facing the same issue as @sunsu123123. Does anyone installed and boot marauder successfully on M5StickC Plus 2?
> I can install it but I can't reboot and the screen stays black for ever... With other firmwares it doesn't happen, so should we wait for a marauder update on this? :/

What you mean by other fws?

devpedromgoncalves, 2023-12-28:
> What you mean by other fws?

Like NEMO or Official. Maybe I used the wrong word, sorry, I'm new to this 😬

kazz2020, 2023-12-28:
> Like NEMO or Official. Maybe I used the wrong word, sorry, I'm new to this 😬

Nemo uses h lib from official m5stack(<M5StickCPlus2.h>) I think there are defined pins for Plus2. Marauder uses Tft_espi lib where we can define for example tft pins. But in code of esp32 marauder we also have defined for example RST pins. In plus there is other RST pin than in plus2. Solo i think that we have to compile ESP32M  fw with this changes. Now I have no time for this but I will see and try I near future 😉maybe I am right. Will see or someone will do that faster than me.

devpedromgoncalves, 2023-12-28:
> > > > Hi. I'm facing the same issue as @sunsu123123. Does anyone installed and boot marauder successfully on M5StickC Plus 2?
> > > > I can install it but I can't reboot and the screen stays black for ever... With other firmwares it doesn't happen, so should we wait for a marauder update on this? :/
> > > 
> > > 
> > > What you mean by other fws?
> > 
> > 
> > Like NEMO or Official. Maybe I used the wrong word, sorry, I'm new to this 😬
> 
> Nemo uses h lib from official m5stack(<M5StickCPlus2.h>) I think there are defined pins for Plus2. Marauder uses Tft_espi lib where we can define for example tft pins. But in code of esp32 marauder we also have defined for example RST pins. In plus there is other RST pin than in plus2. Soooo i think that we have to compile ESP32M fw with this changes. Now I have no time for this but I will see and try in near future 😉maybe I am right. Will see or someone will do that faster than me.

Wow, thanks for the explanation, I'll wait and see if there's any update on this in the near future. Thank you my friend

kazz2020, 2023-12-28:
> Wow, thanks for the explanation, I'll wait and see if there's any update on this in the near future. Thank you my friend

No problem. I don't now if iam right but after new year I will try to do it 😎

gato001k1, 2024-01-03:
> Nemo uses h lib from official m5stack(<M5StickCPlus2.h>) I think there are defined pins for Plus2. Marauder uses Tft_espi lib where we can define for example tft pins. But in code of esp32 marauder [...]

Thanks for the idea i am going to compile it

kazz2020, 2024-01-03:
> Thanks for the idea i am going to compile it

Look carefully for every pin which is different  from plus. I've changed pin numbers 8n source configs.h led pin is different and changed pin numbers in User_setup.h

kazz2020, 2024-01-04:
Ok so I've flashed and no success. [image]

I have this but screen is black. I will check later what could be wrong.

gato001k1, 2024-01-04:
> Look carefully for every pin which is different from plus. I've changed pin numbers 8n source configs.h led pin is different and changed pin numbers in User_setup.h

send the link

gato001k1, 2024-01-04:
> Ok so I've flashed and no success. ![IMG_20240104_090736.jpg](https://private-user-images.githubusercontent.com/44699964/294120874-e6b9a13b-ab22-4601-949f-0d00aa530ecd.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzNzkzOTEsIm5iZiI6MTcwNDM3OTA5MSwicGF0aCI6Ii80NDY5OTk2NC8yOTQxMjA4NzQtZTZiOWExM2ItYWIyMi00NjAxLTk0OWYtMGQwMGFhNTMwZWNkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA0VDE0MzgxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVmZDYyM2NlMTEyZGRkMzc3OTBlODFjNTNhY2E0MmRlNmQzYTBlZGFjMjUyMTM4ODYxMGQ5MWUxZWJjMGRkMDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7O7dK9cVDdxLOPfN5z7wSQGDQrSNa-AuuIpRkHCgp8I)
> 
> I have this but screen is black. I will check later what could be wrong.

damm

kazz2020, 2024-01-04:
> > Ok so I've flashed and no success. ![IMG_20240104_090736.jpg](https://private-user-images.githubusercontent.com/44699964/294120874-e6b9a13b-ab22-4601-949f-0d00aa530ecd.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzNzkzOTEsIm5iZiI6MTcwNDM3OTA5MSwicGF0aCI6Ii80NDY5OTk2NC8yOTQxMjA4NzQtZTZiOWExM2ItYWIyMi00NjAxLTk0OWYtMGQwMGFhNTMwZWNkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA0VDE0MzgxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVmZDYyM2NlMTEyZGRkMzc3OTBlODFjNTNhY2E0MmRlNmQzYTBlZGFjMjUyMTM4ODYxMGQ5MWUxZWJjMGRkMDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7O7dK9cVDdxLOPfN5z7wSQGDQrSNa-AuuIpRkHCgp8I)
> > I have this but screen is black. I will check later what could be wrong.
> 
> Damm is that marauder CLI and have you checked if the display outputs?
Its the output from flashing bin file with ESPHome-Flasher-1.4.0-Windows-x64 
I spent all day to make it work but nothing :/ I've tried many combinations. ST7789_DRIVER and ST7789_2_DRIVER. with mosi 4 pin and without. With different baudrates. Maybe Partition scheme is important? but i tried many. All the time black screen. I give up for now.

gato001k1, 2024-01-04:
> Its the output from flashing bin file with ESPHome-Flasher-1.4.0-Windows-x64
> I spent all day to make it work but nothing :/ I've tried many combinations. ST7789_DRIVER and ST7789_2_DRIVER. with [...]

Idk about that program use spacehun's website esp32 flasher and add the partition scheme and bootloader the boot app and the compiled bin file like he says at   https://github.com/justcallmekoko/ESP32Marauder/wiki/update-firmware#using-spacehuhn-web-updater or give me the link to the compiled bin file

kazz2020, 2024-01-04:
> Idk about that program use spacehun's website esp32 flasher and add the partition scheme and bootloader the boot app and the compiled bin file like he says at [...]

I tried that but no luck.
here is the link for 2 versions
https://file.io/5odwjC4d0V0s

kazz2020, 2024-01-04:
With esp32_marauderm5stickcplus2.ino.m5stack_stickc_plus2 cli is working but still black screen.

kazz2020, 2024-01-04:
I have question. Is there any difference with choosing "Partition scheme" while compiling ESP32M? Is there any difference working/not working? Does the success of the working of fw depend on the choice of scheme?

gato001k1, 2024-01-04:
> I have question. Is there any difference with choosing "Partition scheme" while compiling ESP32M? Is there any difference working/not working? Does the success of the working of fw depend on the choice of scheme? 

I don't know why would you need the partition scheme to compile it. you just need the partition scheme when using it on spacehun's web esp flasher

gato001k1, 2024-01-04:
> I have question. Is there any difference with choosing "Partition scheme" while compiling ESP32M? Is there any difference working/not working? Does the success of the working of fw depend on the choice of scheme? 

Can you send the esp marauder folder that you modified to make it work on the m5plus2

kazz2020, 2024-01-04:
> > I have question. Is there any difference with choosing "Partition scheme" while compiling ESP32M? Is there any difference working/not working? Does the success of the working of fw depend on the choice of scheme?
> 
> Can you send the esp marauder folder that you modified to make it work on the m5plus2

https://file.io/CZTbZC0kJPxt
Make it works :) and when you will dont forget about me ;)

gato001k1, 2024-01-05:
> https://file.io/CZTbZC0kJPxt Make it works :) and when you will dont forget about me ;)

it says deleted

kazz2020, 2024-01-05:
> > > > I have question. Is there any difference with choosing "Partition scheme" while compiling ESP32M? Is there any difference working/not working? Does the success of the working of fw depend on the choice of scheme?
> > > 
> > > 
> > > Can you send the esp marauder folder that you modified to make it work on the m5plus2
> > 
> > 
> > https://file.io/CZTbZC0kJPxt Make it works :) and when you will dont forget about me ;)
> 
> i wont forget but it says deleted

https://filetransfer.io/data-package/nb0JcwQa#link
ive tried many combinations and still nothing :/

kazz2020, 2024-01-05:
pass xxx

kazz2020, 2024-01-06:
[image]
I have this. Only shows on screen when it's plugged in usb. Without usb only blinks Marauder for a sec.

devpedromgoncalves, 2024-01-06:
> ![IMG_20240106_094218.jpg](https://private-user-images.githubusercontent.com/44699964/294660636-c8745dfd-6880-4354-b529-4f8dd2c2c0ff.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MzU4NTEsIm5iZiI6MTcwNDUzNTU1MSwicGF0aCI6Ii80NDY5OTk2NC8yOTQ2NjA2MzYtYzg3NDVkZmQtNjg4MC00MzU0LWI1MjktNGY4ZGQyYzJjMGZmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDEwMDU1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTczZGM1NmNhZDM1NmM1ZmNhOWU0YWYyYzAzN2YxMGZjYmE2ODUxM2VmZTljNjRjMjQ1YjExZmY0NjhjZTNkNGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.O6OMQaceFI8YOVzl1gwBDFAKe4GJzT-sMP4g3ae2hFw) I have this. Only shows on screen when it's plugged in usb. Without usb only blinks Marauder for a sec.

Making progress there! 👏🏻 I've already tried many things but it's really behind my skills for now... Love to see you progress tho!

kazz2020, 2024-01-06:
> Making progress there! 👏🏻 I've already tried many things but it's really behind my skills for now... Love to see you progress tho!

https://file.io/4xbAfvwwLoUg

gato001k1, 2024-01-07:
> ![IMG_20240106_094218.jpg](https://private-user-images.githubusercontent.com/44699964/294660636-c8745dfd-6880-4354-b529-4f8dd2c2c0ff.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY2 [...]

Amazing what did you change to make it work i have tried and i am not able to make the screen display something.

gato001k1, 2024-01-07:
ok now i review the code and when it disconnects the cable it turns off so i checked the code and i am not an expert on this but This doesn't  seems right 😂😂
[image]

kazz2020, 2024-01-07:
https://file.io/s5tzKnonomex
and that 00000 was my 2y old son :P he likes tech ;) ive changed TFT_BL pin to 27. This usb cable issue must be solved. Plus vs Plus2 - plus2 doesnt have axp192 power management chip. But what to change in code for now is mystery for me.

gato001k1, 2024-01-08:
> https://file.io/s5tzKnonomex and that 00000 was my 2y old son :P he likes tech ;) ive changed TFT_BL pin to 27. This usb cable issue must be solved. Plus vs Plus2 - plus2 doesnt have axp192 power management chip. But what to change in code for now is mystery for me.

So what does the m5plus2 does for power management

Jahroots, 2024-01-08:
Hi guys! Happy New Year to you all! Can you tell us what you did to make it work?

gato001k1, 2024-01-09:
> Hi guys! Happy New Year to you all! Can you tell us what you did to make it work?

We changed the display pin configuration and the backlight and i think the led pin but when we compile it it works but if i disconnect the cable and we figured out that is the problem of the power management chip apx192 and we dont know how to fix it.

gato001k1, 2024-01-09:
> https://file.io/s5tzKnonomex and that 00000 was my 2y old son :P he likes tech ;) ive changed TFT_BL pin to 27. This usb cable issue must be solved. Plus vs Plus2 - plus2 doesnt have axp192 power management chip. But what to change in code for now is mystery for me.

Maybe it has something to be with the g38 gpio pin

kazz2020, 2024-01-10:
[image]

WORKING Later I give a link for bin and source

gato001k1, 2024-01-10:
> [image]
> 
> WORKING Later I give a link for bin and source 

What did u change for that and plz bin file link

devpedromgoncalves, 2024-01-10:
> ![IMG_20240110_194929.jpg](https://private-user-images.githubusercontent.com/44699964/295675549-68c35a05-c61b-45ca-8176-0c0432ade6d0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5MjA5NTEsIm5iZiI6MTcwNDkyMDY1MSwicGF0aCI6Ii80NDY5OTk2NC8yOTU2NzU1NDktNjhjMzVhMDUtYzYxYi00NWNhLTgxNzYtMGMwNDMyYWRlNmQwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEwVDIxMDQxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYwN2EwMWZjMjg1YjdiZGVhZGQ0MzY0ZTg5NzVhYzExYmM3ZjJkMTk1NjE4ODEwOTNiYmUwZDczODRmYzJlYTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.k1b2hUXumRz_KAmyfzORe9XBD8A_8suNsM0LMiFItEk)
> 
> WORKING Later I give a link for bin and source

You tha man! Awesome work.

kazz2020, 2024-01-10:
After triggering the wake up signal, the hold(G4) pin needs to be set to a high level (1) in program initialization to maintain the power supply, otherwise the device will enter the shutdown state again.

So I added after setup()

pinMode(4, OUTPUT);
digitalWrite(4, HIGH);

Tomorrow I give link to bin and source.

devpedromgoncalves, 2024-01-10:
> After triggering the wake up signal, the hold(G4) pin needs to be set to a high level (1) in program initialization to maintain the power supply, otherwise the device will enter the shutdown state again.
> 
> So I added after setup()
> 
> pinMode(4, OUTPUT); digitalWrite(4, HIGH);
> 
> Tomorrow I'll give link to bin and source.

I saw that tip on a reddit tbh but never thought about it. Thanks man, will wait for the code. Did you test the firmware? Is fully functional?

kazz2020, 2024-01-10:
> > After triggering the wake up signal, the hold(G4) pin needs to be set to a high level (1) in program initialization to maintain the power supply, otherwise the device will enter the shutdown state again.
> > 
> > So I added after setup()
> > 
> > pinMode(4, OUTPUT); digitalWrite(4, HIGH);
> > 
> > Tomorrow I'll give link to bin and source.
> 
> I saw that tip on reddit tbh but never thought about it. Thanks man, will wait for the code. Did you test the firmware? Is fully functional? 

https://terabox.com/s/1eFejMJ1g-0GBEYIaPjnjww
Now I go to sleep so You can check it by yourself😎

Infotradesolution, 2024-01-11:
https://mega.nz/file/B2FizIQJ#l05BKrCD6Jh8XmtdbodKDcKskiGRMCxSIRuqDtYUB0M

much more easy to download

terabox is a pain in the *ss

WhiskyRed, 2024-01-11:
how to install firmware on m5stack? I'm new to this world, I don't know how to edit code or anything for it

Infotradesolution, 2024-01-11:
> how to install firmware on m5stack? I'm new to this world, I don't know how to edit code or anything for it

No clue 
Wait ppl wakeup n teach
Tried webflasher no Luck 
.zip file? Its src code?

WhiskyRed, 2024-01-11:
I understood and that I am new to the subject and want to learn how to carry out these processes

kazz2020, 2024-01-11:
> how to install firmware on m5stack? I'm new to this world, I don't know how to edit code or anything for it

Use this https://m.youtube.com/watch?v=GQPzgYksdkg and for firmware use bin file from my zip file. After that should work.

WhiskyRed, 2024-01-11:
Na Não consigo conectar meu m5stack ao espwebtool ele consta que devo resetar meu esp mesmo ele usando o firmware demo da m5

kazz2020, 2024-01-11:
> I can't connect my m5stack to espwebtool it says I must reset when resetting my esp even though it is using the m5 demo firmware

Do you use this.   https://esp.huhn.me/

Infotradesolution, 2024-01-11:
Ill give a try later.

devpedromgoncalves, 2024-01-11:
> https://terabox.com/s/1eFejMJ1g-0GBEYIaPjnjww Now I go to sleep so You can check it by yourself😎

Hi again, my friend. I was able to flash marauder onto my M5StickC Plus 2 and it's working fine! 
Now I have another issue. 
When I connect an SD Card Module to it, when rebooting, always say: **Failed to initialize SD Card**. 

I'm using this schema: 
**3V3 > 3V3
CS > GND
MOSI > G26
CLK > G0
MISO > G36
GND > GND.** 

[image]

Is it a problem with my connections or is the code missing some function to read SD Card?

AdonisUchoa, 2024-01-11:
[image]

my is giving this error

WhiskyRed, 2024-01-11:
> > Não consigo conectar meu m5stack ao espwebtool, ele diz que devo redefinir ao redefinir meu esp, mesmo que esteja usando o firmware de demonstração m5
> 
> Você usa isso. https://esp.huhn.me/

give error

WhiskyRed, 2024-01-11:
> > meu está dando esse erro
> 
> Você tem que ter certeza que o seu dispositivo está conectado e está selecionando a porta correta.
> 
> ![imagem](https://private-user-images.githubusercontent.com/97984504/295978952-28a0442a-187e-41da-b5a9-f4648bd9935f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5OTAxMTQsIm5iZiI6MTcwNDk4OTgxNCwicGF0aCI6Ii85Nzk4NDUwNC8yOTU5Nzg5NTItMjhhMDQ0MmEtMTg3ZS00MWRhLWI1YTktZjQ2NDhiZDk5MzVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTExVDE2MTY1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5ZWRhMzAxZTExYTc5Y2I1MzRkMTUzZWZiYWFhZDVhOGRiNGQ0YzliZDRjMWI5NTdhODVlYWI3MjlmZmJlYzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.A7uAQnbpxmTWzp92r2grilRyNhB5rOPV3ro__LPpsVs)

[image]
I'm using com3 and I'm sure it's him

Infotradesolution, 2024-01-11:
Mine worked 
Marauder running fine
No Black screen
No errors 

Tks folks

AdonisUchoa, 2024-01-11:
> > > meu está dando esse erro
> > 
> > 
> > Você tem que ter certeza que o seu dispositivo está conectado e está selecionando a porta correta.
> 
> ![image](https://private-user-images.githubusercontent.com/156268711/295980545-63c88aae-b255-4791-b0ff-8afb037acc82.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5OTE2MzgsIm5iZiI6MTcwNDk5MTMzOCwicGF0aCI6Ii8xNTYyNjg3MTEvMjk1OTgwNTQ1LTYzYzg4YWFlLWIyNTUtNDc5MS1iMGZmLThhZmIwMzdhY2M4Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMVQxNjQyMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NGFmYzk1NGFiZjY0YmVkNWRmNWYyOTU2OWZmMjk2Zjg0NGU2ZGNlY2UyZGE2YzJmNzk4YjQ0YWUxNzczOGFjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.feAgJ5sQQiJohyndjVCeonBYX_mv6i37m9WDSly65Lw) I'm using com3 and I'm sure it's him, I don't know what to do, I've tried everything and nothing

it was not recognizing it because there was a drive preventing COM communication

WhiskyRed, 2024-01-11:
> O meu funcionou Marauder correndo bem Nenhuma tela preta Sem erros
> 
> Tks pessoas

How did you do the installation?

[38 more replies]

closed 2024-02-03 by kazz2020