justcallmekoko/ESP32Marauder #440: Compilation error with XIAO ESP32S3 board

VNEvil, 2023-12-21:
Hi @justcallmekoko 
When I compile your ESP32Marauder source code, I encounter a series of errors as follows (I'm using the Windows 10 operating system and have tried compiling on Windows, Linux, Visual Studio Code but all received the same result):
```
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(Buffer.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:189: multiple definition of `text_table4'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:189: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(Buffer.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:188: multiple definition of `text_table3'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:188: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(Buffer.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:187: multiple definition of `text_table2'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:187: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(Buffer.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:186: multiple definition of `text_table1'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:186: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(Buffer.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:185: multiple definition of `text_table0'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:185: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(CommandLine.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:189: multiple definition of `text_table4'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:189: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(CommandLine.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:188: multiple definition of `text_table3'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:188: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(CommandLine.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:187: multiple definition of `text_table2'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:187: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(CommandLine.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:186: multiple definition of `text_table1'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:186: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(CommandLine.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:185: multiple definition of `text_table0'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:185: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(Display.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:189: multiple definition of `text_table4'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:189: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(Display.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:188: multiple definition of `text_table3'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(BatteryInterface.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:188: first defined here
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(Display.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/lang_var.h:187: multiple definition of `text_table2'; C:\U [... cut ...]
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; C:\Users\Anonymous\AppData\Local\Temp\arduino\sketches\455BF3969E8E6B7F77EBAD28E46FAD21\sketch\objs.a(WiFiScan.cpp.o):C:\Users\Anonymous\Videos\ESP32Marauder-0.13.6\esp32_marauder/WiFiScan.cpp:13: first defined here
collect2.exe: error: ld returned 1 exit status

```
I have tried to fix it for many days but have not received any feasible results.
Can you tell me the reason why such an error occurred?
Thank you very much!!!"

XANKXANK, 2023-12-21:
maybe try asking kashmir54 he did it on xiao seeed 32c3 maybe he can help you

smoochiee, 2023-12-23:
https://github.com/justcallmekoko/ESP32Marauder/wiki/faq

FrancoSpermasolido, 2024-07-25:
did you find a solution for this?

VNEvil, 2024-07-25:
You need to modify the **platform.txt** file in arduino to support the XIAO ESP32S3 board. You need to follow these steps:
1. Open the **platform.txt** file with any text editor at the following path:
`C:\Users\<your_username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\platform.txt`
2. Proceed to modify the contents of the **platform.txt** file:
- Find and add **-w** to the **build.extra_flags.esp32s3** option:
`build.extra_flags.esp32s3=**-w** -DARDUINO_USB_MODE={build.usb_mode} -DARDUINO_USB_CDC_....`
- Find and add **-zmuldefs** to the **compiler.c.elf.libs.esp32s3** option:
`compiler.c.elf.libs.esp32s3=**-zmuldefs** -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_up....`
Wishing you success!

FrancoSpermasolido, 2024-07-25:
> You need to modify the **platform.txt** file in arduino to support the XIAO ESP32S3 board. You need to follow these steps:
>
> 1. Open the **platform.txt** file with any text editor at the following [...]

I just did, sadly i still get the same result, i have linked the error code.
[Nuovo documento di testo (2).txt](https://github.com/user-attachments/files/16380587/Nuovo.documento.di.testo.2.txt)