justcallmekoko/ESP32Marauder #442: Evil Portal M5StickC Plus

jedd2169, 2023-12-21:
As I click on Evil Portal it says "Could not find /ap.config.txt.
Touch to exit ..." I see you have to set it up bit you need a sd card is there anyway around this for the m5stickc plus?

Masterbudz01, 2023-12-23:
i have connected a sd card module reader to m5stick plus but when sd load m5stick restarts

CavaleiroCinza, 2023-12-31:
> i have connected a sd card module reader to m5stick plus but when sd load m5stick restarts

Hey man, so... the new versions like 13.5/13.6 are not working with the SD card module on the m5! We'll have to wait for updates... now about the evil portal, what are these versions 
Older versions do not have the evil function enabled, they are missing a code so that they can read the card and select some files. These codes were added in versions 13.5/13.6 but as they are not available
We'll have to wait for updates lol!

CavaleiroCinza, 2023-12-31:
> As I click on Evil Portal it says "Could not find /ap.config.txt. Touch to exit ..." I see you have to set it up bit you need a sd card is there anyway around this for the m5stickc plus?

Hey man, so... the new versions like 13.5/13.6 are not working with the SD card module on the m5! We'll have to wait for updates... now about the evil portal, what are these versions 
Older versions do not have the evil function enabled, they are missing a code so that they can read the card and select some files. These codes were added in versions 13.5/13.6 but as they are not available
We'll have to wait for updates lol!