justcallmekoko/ESP32Marauder #444: Empty PCAP after flash the sd_serial.bin

kajlo, 2023-12-23:
Hi, is anyone has an idea, why i getting empty (0byte) PCAP files ?
Saving logs are fine, its also create the PCAP files, but they empty.

I using the Newest Xtreme firmware (XFW-0052) with a Wifi dev board
I used FZ_Marauder Flash 2.5 with the newest marauder firmware.
I change the bin file in Marauder folder to sd_serial.bin. 
Flashed it, and it doesnt work.
It used to be work for me, a few month back. Unfortunately i cant remember what version.

Is anyone has any idea about this ?
also i noticed when i open marauder application on flipper zero, there is a "LIST SD" menu, when i press it, it says 'SD support disabled, cannot use command'

hadrianobrasil, 2023-12-24:
Hey bro, same problem here. Its a bug on xtreme 52. Just downgrade to xtreme 51 and be happy again :)

kajlo, 2023-12-24:
> Hey bro, same problem here. Its a bug on xtreme 52. Just downgrade to xtreme 51 and be happy again :)

Its worked. Cheers mate, i tried a few different marauder firmware, but actualy i never think to change the flipper firmware.

closed 2023-12-24 by kajlo