justcallmekoko/ESP32Marauder #451: Problems with SD card on M5STICK C PLUS

CavaleiroCinza, 2023-12-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
New versions for M5STICK C PLUS do not work.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
When installing new versions such as 13.5/13.6, they themselves cannot recognize the SD card, so you will only be able to use marauder using older versions.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Even older versions are problematic, such as: I can't write to the SD card and the text is too small. The solution is to fix all new versions for the M5.
**Additional context**
Add any other context or screenshots about the feature request here.

Masterbudz01, 2023-12-30:
i have tried with 13.3 version and all obsolete apps works, but when i start evil portal, device crash and restarts disconnecting SD card.. i don't know why.

CavaleiroCinza, 2023-12-30:
> i have tried with 13.3 version and all obsolete apps works, but when i start evil portal, device crash and restarts disconnecting SD card.. i don't know why.

Again like I said, it's a programming problem. Either we do the correction at home or wait for JustCallMeLoko to correct it; In the end we have to wait for updates!

closed 2026-10-11 by github-actions[bot]