justcallmekoko/ESP32Marauder #460: Screen

Pipinice, 2024-01-05:
Hi,
Does anyone know why the screen is backwards on esp32 tft 2.8? 
All connections are correct.

[image]
[image]
?

gato001k1, 2024-01-06:
maybe the size of the screen is different from the compiled bin file

Pipinice, 2024-01-06:
The size is correct, it is 2.8 It was these files that I programmed, I also tried it with v6 and I do the same.

[image]

Pipinice, 2024-01-06:
Solved

CheekiBreeki95, 2024-01-07:
How did you solve it?

Pipinice, 2024-01-07:
Send a photo of what the screen looks like. What screen do you use?

CheekiBreeki95, 2024-01-08:
Exact same screen, exact same issue

Pipinice, 2024-01-08:
Try this files and report.

https://we.tl/t-9KWL2OmeaG

CheekiBreeki95, 2024-01-08:
Could you just tell me what you changed?

CheekiBreeki95, 2024-01-08:
Nevermind its fixed. For anyone looking at this with the same problem you HAVE to build from source for the screen to work on a DIY model.

CheekiBreeki95, 2024-01-09:
What?

Pro123123kk, 2024-01-11:
it has a blank screen error :(
[image]

An0mXD, 2024-01-11:
I use the same screen as you but the touchscreen doesn t work and i can t use it
please , help me

CheekiBreeki95, 2024-01-13:
Pro123123kk your image doesnt work
An0mXD check your wiring and try using the stylus

Pro123123kk, 2024-01-18:
The link above to fix that screen has expired, can you resend it? 
Pipinice

Pro123123kk, 2024-01-18:
[image]

CheekiBreeki95, 2024-01-18:
> The link above to fix that screen has expired, can you resend it? Pipinice

Just build it yourself, looking at the new issue you opened your screen is different so it might need a different config, follow the guide at [this link](https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup) to setup the IDE then the guide at [this link](https://github.com/justcallmekoko/ESP32Marauder/wiki/installing-firmware-from-source) to build it

R0dr1g0r, 2024-01-21:
> Try this files and report.
> 
> https://we.tl/t-9KWL2OmeaG

can you send this again pls

CheekiBreeki95, 2024-01-21:
> I'm not good at this, can you give me the bin file to flash?

Like I said in the my previous message you'll have to build it yourself, its rather straightforward.

R0dr1g0r, 2024-01-21:
> > I'm not good at this, can you give me the bin file to flash?
> 
> Like I said in the my previous message you'll have to build it yourself, its rather straightforward.

humm okok I'm following the tutorial on how to modify the firmware right now thx :)

KingJoke1990, 2024-02-14:
@An0mXD Im having the same issue as you. Were you able to fix the problem?

hadrianobrasil, 2024-02-28:
> Solved

can u tell us how ? please

same problem here

FrancoSpermasolido, 2024-08-03:
any updates on the issue?

QuangVuDEV12, 2024-09-12:
Your screen is ST7789 driver, so you need ili9341 tft or use marauder for st7788

closed 2025-05-30 by justcallmekoko