justcallmekoko/ESP32Marauder #465: Error with SoftwareSerial.h

BirdGGbro, 2024-01-21:
**I get this error:**
In file included from C:\Users\Lyric\Downloads\ESP32Marauder-0.13.6\ESP32Marauder-0.13.6\esp32_marauder\esp32_marauder.ino:28:
C:\Users\Lyric\Downloads\ESP32Marauder-0.13.6\ESP32Marauder-0.13.6\esp32_marauder\GpsInterface.h:7:10: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Multiple libraries were found for "SD.h"
  Used: C:\Users\Lyric\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\SD
  Not used: C:\Users\Lyric\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: SoftwareSerial.h: No such file or directory

**then after installing the library I get this error:**
C:\Users\Lyric\AppData\Local\Arduino15\libraries\SoftwareSerial\src\SoftwareSerial.cpp:41:10: fatal error: avr/interrupt.h: No such file or directory
 #include <avr/interrupt.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

chmod771, 2024-01-28:
What board are you targeting?

BirdGGbro, 2024-01-29:
I use the ESP32-WROOM-32U board, but in the Arduino program I set LOLIN D32

Ryan-py, 2024-02-04:
@BirdGGbro  i had the same problem but turns out for esp32 boards you have to install  `EspSoftwareSerial`

BirdGGbro, 2024-02-04:
@Ryan-py thank you, I will try later and if it works, I will be very grateful to you

Axel2609, 2024-12-07:
> @BirdGGbro i had the same problem but turns out for esp32 boards you have to install library`EspSoftwareSerial`

And this is the correct answer! Thanks!