justcallmekoko/ESP32Marauder #471: S3 bluetooth

SomeToms64, 2024-02-04:
S3 bin has bluetooth disabled due to using the same settings as the S2.
The S3 has bluetooth so i dont see why it wouldnt be enabled.

detonation, 2024-02-17:
I also just noticed this. Previous version DID have bluetooth enabled, so it was probably overlooked when they rolled out the -serial change of 13.7. I cannot recall 100% if it was 13.5 or 13.6 s3 that did have BT enabled, but 13.7 s3 has it disabled. Which is disappointing. Perhaps they will fix it the next version, otherwise just compile it from source

avipars, 2024-12-24:
> Hey there,
> 
> Here is a bin file for the S3 compiled by us with the BL enabled: [ElectronicCats/flipper-addons#4](https://github.com/ElectronicCats/flipper-addons/issues/4)

Any chance of merging it?

closed 2026-10-20 by github-actions[bot]