justcallmekoko/ESP32Marauder #498: better ble spam

ISoloRaid, 2024-02-24:
samsung and google ble spams just shows only 2 pop ups and stop, you can send more 2 pop ups after a time, so inconsistent. And the Sourapple dont works more after ios 17.3 update, add applejuice instead (but the swiftpair works greatly)

namara624, 2024-02-25:
Same goes for me i was lookin to see if logs would show anything. but it spams the devices with BLE and when i go to "confirm" or even see it on my device, i now see that the Marauder is power cycle.

copyrightsclient, 2024-03-09:
Hello, I want to use applejuice, but I do not know how to install from the marauder source file. Can you help me?
@ISoloRaid 
Discod: sorgu_paneli
mail: rwkvnx@gmail.com

closed 2026-10-10 by github-actions[bot]