justcallmekoko/ESP32Marauder #504: Touch screen issue

Bryplak, 2024-02-28:
https://github.com/justcallmekoko/ESP32Marauder/assets/96693780/81bf7608-706e-4f6d-9c9e-1f87364e1c2dUsing v4 my touch screen was inverted. I flashed v6 but now the top left of screen is fine but to click on lower portion of menu I have to touch the top right of screen.

Makss8, 2024-03-01:
Have same problem

indavac, 2024-03-03:
I have a similar issue, can you fixed?

https://github.com/justcallmekoko/ESP32Marauder/assets/36338090/2c6920f4-5660-44c1-b3ad-f32cdd5b6a44

Bryplak, 2024-03-04:
[indavac](https://github.com/indavac) comment try v6 not v4

Bryplak, 2024-03-04:
as for my issue it looks like AliExpress sent me the wrong size display.
they sent 2.4 instead of 2.8

closed 2024-03-04 by Bryplak

indavac, 2024-03-04:
> [indavac](https://github.com/indavac) comment try v6 not v4

thanks dude, you fixed!