justcallmekoko/ESP32Marauder #509: Marauder NOT Sniffing the wifi network I selected, but a RANDOM other wifi network...

Dochartaigh, 2024-03-08:
**Describe the bug**
WiFi Marauder program on Flipper Zero is scanning the WRONG WiFi network when you're doing something like Sniff (with "pmkid" option). And yes, of course I double checked I selected the right network, and it is definitely STILL selecting the wrong WiFi network (and giving me wrong PCAPs since they're for a different WiFi network than I selected).

**To Reproduce**
Steps to reproduce the behavior:
1. Open WiFi Marauder program on your Flipper Zero with official WiFi Dev Board Attached (flashed with Marauder).
2. Choose "Scan (ap)" option and let it run for a bit, then back out.
3. Choose "List (ap)" find the number of the network you want to Sniff, then back out.
4. Choose "Select (ap)" and add the number of the network you saw in the list above, hit "Save", back out.
5. Choose "Sniff (pmkid)" and let it run until you get PCAP files (it'll say "Received EAPOL"). In those PCAP files you'll see it's for the WRONG network - it scanned some other random WiFi network in your area. It did NOT scan the network you selected (and double-checked that you selected is correctly). This doesn't happen 100% of the time, just a lot!

**Expected behavior**
I expect WiFi Marauder to scan the network I selected, and not another random network.

**Screenshots**
[image]

**Marauder (please complete the following information if applicable):**
 - Firmware version: Marauder 0.13.9
 - Hardware version: OG/Official Flipper Zero WiFi Dev Board (ESP32-S2-WROVER based) 
 - Flipper Zero running newest XFW Firmware, OR newest RogueMaster firmware - happens on BOTH.

**Additional context**
Just started researching but at least one other person (with same hardware as me, on newest versions of everything) has this SAME EXACT Issue! ...so at least it's not just me ;)

intentethan, 2024-03-08:
I have the same issue 13.9

Dochartaigh, 2024-03-08:
> I have the same issue 13.9

Are you using a Flipper Zero with official WiFi Dev Board (flashed with Marauder)? If so, what Flipper firmware are you on? ...don't know if this GitHub is mostly flipper people or what which is why I asked (and also want to see if it's a flipper-only thing, or effects anybody with a ESP32 based WiFi board running Marauder – on whatever other type of non-flipper hardware).

MastiffJeff, 2024-03-08:
Same issue with me, Flipper 13.9 and official dev board.

Dochartaigh, 2024-03-08:
Have either of you @intentethan @MastiffJeff rolled back the Marauder firmware and see if earlier versions work properly? I just got my Flipper this week so only been using the newest version. 

Also wanted to ask you both how fast yours gets the PCAP file (where you run "Sniff (pmkid)", and it completes by saying "Received EAPOL"). Every. single. video I watch they seem to get the "Received EAPOL" message super fast (could be the editing though?)... Mine does NOT work this way. Many times I can run it for an HOUR (if not several) and it won't get one. Sometimes I'm lucky and can hit back button and try it again and it might get one immediately, or in a ~minute i.e. super fast... but many times it's the same and nothing for a very long time (if not back out and try again multiple times over and over again... can commonly take FOREVER). -- just wanted to make sure this is normal... if not it might be related to this issue ("Sniff (pmkid)") which is why I mentioned it.

intentethan, 2024-03-08:
i do also have this happen to me but only on some networks, i have not rolled back

MastiffJeff, 2024-03-09:
I have not rolled back yet.

InfoSecREDD, 2024-03-09:
This also happens when the network you're targeting has other networks on the same channel, my guess is the firmware is looking at the channel rather than the SSID..

InfoSecREDD, 2024-03-09:
@Dochartaigh please close the issue, we have linked the solution to it in Sasquach's Discord.

Dochartaigh, 2024-03-09:
Just so people know the solution: In Talking Sasquach's (very popular) 2024 video on WiFi and Marauder, when he's going over the process in the WiFi Marauder app, in the "Sniff (pmkid)" options, he mistakingly says to choose "Active (ForceDeauth)". This is incorrect – you want to use the "Targeted Active (List)" option. That's the one which should scan ONLY the network/SSID you have selected. 

...have a feeling with that video being so popular, and so many people reporting this as an issue, probably quite a few watched that same video and are making the same mistake.

MastiffJeff, 2024-03-09:
I just tested this, target on channel 3, used Targeted Active PCAP file was my home router on channel 10.

Dochartaigh, 2024-03-09:
Sorry to re-open but I just did some more testing with "Targeted Active (List)"... and just like when I was using "Active (Force Deauth)", the PCAP I got was for the WRONG network again! ...so I think this is still broke.

On the "Flipper Zero Level Up" Facebook group (where somebody posted about this GitHub ticket) another user posted saying he ALWAYS uses "Targeted Active (List)" from the beginning and he likewise gets PCAP's for the wrong network so that's another person confirming this... so this leads me to believe this bug effects multiple Sniff (pmkid) options.

Also wanted to note that for anybody trying to duplicate this bug I think you HAVE to be in a very Wi-Fi heavy area. We think it's scanning EVERYTHING on a single channel to grab a PCAP - so the PCAP you get back could be the network you selected on that channel, but it could also be some other RANDOM network on that SAME Channel. When I "Scan (ap)" on mine, after 30 seconds there's 50+ SSID's listed, with TONS on the same channel... (just in case somebody posts saying theirs works fine)

ph3on1x, 2024-08-19:
Same here. Momentum FW, WiFi Dev board, flashed through ESP32 Flasher included with FW. Tried different ways (reflash, reinstall FW, other FWs etc), still same problem - with targeted (active/passive) wrong ap is taken. It seems there is a mismatch with ap list ids provided by Scan ap.

ChristopherJWallace, 2024-09-09:
Same issue for me. Momentum FW, WiFi Dev board, flashed through ESP32 Flasher included with FW. 

Viewed list and selected the correct ID and noted the ch#. Sent Sniff (pmkid) active (Force Deauth) command to ID 0 channel #1 and it sent to channel #6.

bir7822, 2024-10-23:
I have the same problem! Has anyone found a solution?

rdellosa, 2024-11-01:
@Dochartaigh I'm having the same issue. Have you found a solution yet?

Topocoje, 2024-11-10:
some1 can help? there is still 0 eapol in pcaps :F

kanekar, 2024-12-01:
Having the same issue. It tells me I'm sniffing 11 but selecting 17

injcristianrojas, 2024-12-27:
Same problem here. When selecting an AP on the list, the channel that sniffpmkid shows when doing a passive scan is different than the one that I selected originally on my list.

Versions:
* ESP Marauder 1.2.0
* Flipper Zero Firmware: mntm-008

fluffysnaff, 2025-01-16:
Hello, I had same issue.
Versions:
- flipper wifi devboard(1.2.0 marauder firmware), no sd mod
- flipper firmware: mntm-008

I was trying this on my 2g wpa/wpa2 network like this:
1. Scan for APs
2. Select AP
3. Sniff PMKID
4. Targeted Active(List)

The channel it said it's sniffing on didn't match the one in the `List` cmd, but after restarting the sniff it started deauthorising on the correct AP. My only issue was I couldn't see EAPOL protocol when opening the .pcap file even though I got `Received EAPOL` on the flipper. 

When opening the .pcap file I got the `The capture file appears to be damaged or corrupt.` msg, but after I used pcapfix I could see the EAPOL protocol packets and the error disappeared.

[image]

In the pcapfix log it found corrupted and overlapping files.
Hope this helps

cbytestech, 2025-02-05:
So I figured out a way...

Sniff PMKID > On Channel # - Active

I was running into the same with seeing my network on ch 1, but yet it would scan other channels(different each scan) with no eap/eapol in pcap.

I ran this way(in video) and it worked almost immediately. no intentional disco/reco to network

[Google Photos Album: Github repo](https://photos.app.goo.gl/74f7dgxow57T2KFS6)

Dochartaigh, 2025-02-05:
> So I figured out a way...
> 
> Sniff PMKID > On Channel # - Active
> 
> I was running into the same with seeing my network on ch 1, but yet it would scan other channels(different each scan) with no eap/eapol in pcap.
> 
> I ran this way(in video) and it worked almost immediately. no intentional disco/reco to network
> 
> [Google Photos Album: Github repo](https://photos.app.goo.gl/74f7dgxow57T2KFS6)
> 
> Just got my flipper for christmas from the wifey just before wedding. Love her, and I love it!
> 
> Hope this helps someone

How many networks do you have by you? I ask because I have about 100 or so (dense area with small houses close together)... I tried it this way before and the same problem occurred.

cbytestech, 2025-02-06:
currently 10, but I believe some are turned off during the day as eveningsit jumps to about 30 in this loft complex

ph3on1x, 2025-02-06:
Problem still exist with latest momentum fw. No issues with other custom fw's, seems to be momentum specific.

cbytestech, 2025-02-06:
ah. yea Im using stock, then just flashing the card w marauder
*-Nicholas 'Norm' Hess*

dvanmatre, 2025-02-09:
> Problem still exist with latest momentum fw. No issues with other custom fw's, seems to be momentum specific.

Found this looking for a fix. Noticed the same thing as @injcristianrojas described. Searched discord, doesn’t seem to be much awareness about this from momentum.

davidequirillo, 2025-03-16:
Hello, 

in my opinion this strange thing happens because the marauder firmware code must be concise and efficient for all possible devices, so the behavior can be different from the one expected reading the menu items.

In the flipper (unleashed firmware, but probably in "momentum" too) the behavior of "Sniff PMKID" is the following:

- PASSIVE: it sniffs all eapol packets related to all access points in the current wifi card channel (note: before start the sniff, go to channel->set, then you choose the desired channel, and then you start passive pkmid sniff). The channel will remain fixed. It will not change during the sniffing. 
Another **Important warning**: in the settings, you must "disable" Force PMKID (and for a total safety disable ForcePROBE too, if you usually use probe sniff), otherwise this sniff become active, with the risk of deauthentication of your neighbors that operate in the same channel.

- ACTIVE: the same of passive, but injects deauth packet in the channel, to deauth all access points/stations that operate in that channel, with the aim of facilitating eapol sniffs

- TARGETED PASSIVE (list): the same of passive. I think it's not targeted at all. But remember to disable "Force PMKID" in the settings, to avoid deauth packet injections, that would disturb the neighbors on the same channel.

- TARGETED ACTIVE (list): in this case, the injected deauth packet will be related only to the selected access points (so it's more safe, so you don't disturb neighbors, if you select only your AP for example). 
But the curious thing is that the eapol packets are sniffed from all channels, because in the actual version of marauder (1.2.1), here there is channel hopping. In other words, the current wifi card channel continues to change, to rotate (as it also happens in "scanAP" for example). This behavior is visible if you analyze source code, in the function wifi scan main().
In my opinion it would be better to keep the channel fixed as in the previous cases, for a better accuracy of the sniffing, and to deauth only the selected AP that operate in the fixed desired channel. I'll try to pull a request to the author probably to do this update in the code.

- CHANNEL PASSIVE: the same as passive, in the desired channel (as input argument)
- CHANNEL ACTIVE: the same as active, in the desired channel (as input argument)

NightfallChease, 2025-06-11:
Same problem here. I have to set the channel manually for sniffing and when I want to scan for signal strenght, it always selects a random channel, so there's no way for me to get the signal strenght scan on the same channel as my targeted wifi.

I thought it was the app that messed up the channels until I saw that the wifi marauder manages the selection etc.

DevBoard: Marauder v1.7.0
Flipper: unlshd-081

closed 2026-10-18 by github-actions[bot]