justcallmekoko/ESP32Marauder #519: Extract Wi-Fi passphrase from AP via known WPS PIN

voiduser123321, 2024-04-07:
The title says it all, basically. It would be something analogous to reaver's -p option. Do you think it would be feasible to implement?

MisterFenek, 2025-05-18:
yeah i think it would be cool, i think it's possible because while scanning, it rarely shows "RXd WPS config of [network]" or smh like that

voiduser123321, 2025-05-20:
> yeah i think it would be cool, i think it's possible because while scanning, it rarely shows "RXd WPS config of [network]" or smh like that

It's a shame that the owner of repo apparently stopped paying attention to this repo's issues. It doesn't matter to me, though, because I abandoned the idea of using Flipper+ESP32 Board as a Wi-Fi pentesting tool, for all it's benefits is outweighed by one huge con - it doesn't support 5/6 Ghz band. You're will be much better off compiling the driver for the external dongle for LineageOS. Not to mention the ability to use an external antenna with it out of the box.

closed 2026-10-18 by github-actions[bot]