justcallmekoko/ESP32Marauder #520: Increasing the user experience

SuthaharanTharunjan, 2024-04-09:
I like your work it is a good pen testing tool it is good but it has one issue. It does not have an friendly gui to communicate with the user I think you are not using the true potential of the tft display 
I know it is easy to tell but hard to do 
Your project is functional but not user friendly.  
I think you can extend the capabilities of the device by adding nrf module and nfc or rfid module
Thank you for the great project.

meow2x, 2024-04-09:
> I like your work it is a good pen testing tool it is good but it has one issue. It does not have an friendly gui to communicate with the user I think you are not using the true potential of the tft [...]

I might agree with the user experience, not really. I like the fact that koko is laser focused on something and making it the best possible device of it's kind. Taking a stray path from the whole point of the project = less time and innovation for the whole purpose. If there are gui/ux masters out there who are willing to help i get it however losing focus from engineering capability disagree. -mm

SuthaharanTharunjan, 2024-04-09:
I know kolo is trying make it good as possible. I just saying that improving ui and adding extra features also a kind of improvement.
I feel that device is missing something in its look it is not provisional enough and tft display's potential is not fully used here. I know it is hard to improve it on its look.
By increasing gui and features it can gain more people's attention like flipperzero.

closed 2026-10-10 by github-actions[bot]