justcallmekoko/ESP32Marauder #533: Doing a signal monitor on a Flipper Zero doesn't show RSSI values

Mr110398, 2024-04-27:
Using the AP list and signal monitor does not show the RSSI value.

[image]

However, when connecting via CLI from PuTTY or Serial USB Terminal app on Android, an unknown character is displayed.

PuTTY:
[image]

Serial USB Terminal app:
[image]

 I'm not a programmer, but I was told that it could be that the RSSI value is being displayed as an ASCII code, but I honestly have no idea.

cul8rmom1, 2024-05-08:
Same here.  Seems broken.  I am using esp32 wifi marauder, using momentum fw latest release today.

davidd227, 2024-05-27:
Same here. I am also using esp32 wifi marauder, with momentum fw latest version. Had the same problem with Xtreme Fimware XFW-0053_02022024 and now switched to momentum with the same problem.

cul8rmom1, 2024-06-27:
Same issue running unleashed firmware

nickytoothiccy, 2024-07-17:
Tried today on Unleashed FW 075 and have the same issue. The values need to be decoded, I wrote a script that does that and it worked for me
[Maruader_log.log](https://github.com/user-attachments/files/16266918/Maruader_log.log)

nickytoothiccy, 2024-07-17:
I believe I found a potential solution. I tried today on Unleashed FW 075 and got the same result. The log needs to be decoded to get the correct values. I used the script attached and it converted the unknown characters.

def decode_rssi(encoded_char):
    return ord(encoded_char) - 256

# Read the log file
with open('marauder_log.txt', 'r', encoding='iso-8859-1') as file:
    lines = file.readlines()

# Process and print each line
for line in lines:
    if 'RSSI:' in line:
        parts = line.strip().split('RSSI:')
        ssid = parts[0].strip()
        encoded_rssi = parts[1].strip()
        decoded_rssi = decode_rssi(encoded_rssi)
        print(f"{ssid} RSSI: {decoded_rssi} dBm")
    else:
        print(line.strip())

cul8rmom1, 2024-10-15:
Mmm kay…. So how do i make it do the do more gooder?

closed 2026-10-18 by github-actions[bot]