justcallmekoko/ESP32Marauder #535: Marauder v4 stuck in white screen

Phat29072011, 2024-05-02:
**Describe the bug**
A clear and concise description of what the bug is.
Marauder stuck in white screen

**Expected behavior**
my serial monitor say this:
20:05:55.725 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:05:55.725 -> configsip: 0, SPIWP:0xee
20:05:55.725 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:05:55.725 -> mode:DIO, clock div:1
20:05:55.725 -> load:0x3fff0030,len:1100
20:05:55.725 -> ho 0 tail 12 room 4
20:05:55.725 -> load:0x40078000,len:12308
20:05:55.725 -> load:0x40080400,len:3076
20:05:55.725 -> entry 0x400805ec
20:05:56.264 -> E (539) esp_core_dump_flash: N⸮⸮

**Marauder
 - Firmware version: [v0.9.17]
 - Hardware version: [OG] 

**Desktop 
 - OS: [Windows 11]
 - Browser [Google Chrome]
 - Version [v0.13.10]

Phat29072011, 2024-05-02:
Help me pls!

LTQ2022, 2024-05-04:
一样

Atonic85, 2024-10-19:
Have you added the display driver or edited the user_setup.h

JLowe269, 2025-01-28:
Any fix for this one @Phat29072011?

closed 2026-10-18 by github-actions[bot]