justcallmekoko/ESP32Marauder #539: a list of all the html files for evil portal instead of just using index.html

CyberTechNex, 2024-05-10:
Hi, 
I was wondering if there could be a feature to choose a html file for evil portal from the files present on the sd card instead of just picking the file named index.html.

I've seen this on the flipper and was wondering if it would be possible on the esp 32 v4.

ATOMNFT, 2024-05-11:
It is possible. I'm not sure why you can't choose other HTML files. What are running marauder on?
[image]

CyberTechNex, 2024-05-11:
I am using a normal esp 32 wroom with the ili9341 display. When I don't rename any file it just shows index.html not found

justcallmekoko, 2024-05-11:
If you don't select any of the HTML files from this list shown in the picture, it defaults to "index.html"

closed 2024-05-11 by justcallmekoko

CyberTechNex, 2024-05-11:
Where do I select the html page in the menu, cause I just select the evil portal option.

ATOMNFT, 2024-05-13:
> I am using a normal esp 32 wroom with the ili9341 display. When I don't rename any file it just shows index.html not found

Have you tried to go into "select EP HTML file" ?

[image]

CyberTechNex, 2024-05-13:
I did not have the option before , maybe i was on an older version. I reflased using the arduino code and now everything works.

ATOMNFT, 2024-05-13:
> I did not have the option before , maybe i was on an older version. I reflased using the arduino code and now everything works.

Glad you got it working!