justcallmekoko/ESP32Marauder #553: Missing _multiboardS3.bin binary for release v1.0.0

maxromanovsky, 2024-06-29:
Missing _multiboardS3.bin binary for release v1.0.0

Could you please re-build/attach it to the release?

[image]

maxromanovsky, 2024-07-05:
@justcallmekoko any chance to add the binary?

kokein, 2024-07-31:
Maybe it is no longer supported for some reason? @justcallmekoko ?

donniebrasc, 2024-09-15:
I like S3's too...

haojiahuo123, 2024-09-16:
Mine is also a
[image]
n S3, but there is no corresponding firmware. Do I need to buy a new S2?

closed 2026-10-18 by github-actions[bot]