justcallmekoko/ESP32Marauder #555: Dont compile for Generic_ESP32 (ESP32-C3)

rbuzatti, 2024-07-10:
The lastest ESP32Marauder, downloaded from git 09 of July, isnt compiling for Generic_ESP32 boards, even using version 1 or 2 from Arduino IDE.

All the libraries are from WiKi, ipsis literis, even not in their latest versions.

The SoftwareSerial.h was downloaded from the Internet and manually copied to the sketch folder.

I used two different computers to try.... One with Windows 10 and other with Windows 11
 
The ERROR message from Arduino IDE was:

C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp: In member function 'void WiFiScan::executeSwiftpairSpam(EBLEPayloadType)':
C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp:1921:5: error: 'esp_base_mac_addr_set' was not declared in this scope; did you mean 'esp_base_mac_addr'?
 1921 |     esp_base_mac_addr_set(macAddr);
      |     ^~~~~~~~~~~~~~~~~~~~~
      |     esp_base_mac_addr
C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp: In member function 'void WiFiScan::RunBluetoothScan(uint8_t, uint16_t)':
C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp:2366:17: error: 'class NimBLEScan' has no member named 'setAdvertisedDeviceCallbacks'
 2366 |       pBLEScan->setAdvertisedDeviceCallbacks(new bluetoothScanAllCallback(), false);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp:2403:19: error: 'class NimBLEScan' has no member named 'setAdvertisedDeviceCallbacks'
 2403 |         pBLEScan->setAdvertisedDeviceCallbacks(new bluetoothScanAllCallback(), false);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp:2405:19: error: 'class NimBLEScan' has no member named 'setAdvertisedDeviceCallbacks'
 2405 |         pBLEScan->setAdvertisedDeviceCallbacks(new bluetoothScanAllCallback(), true);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp:2422:17: error: 'class NimBLEScan' has no member named 'setAdvertisedDeviceCallbacks'
 2422 |       pBLEScan->setAdvertisedDeviceCallbacks(new bluetoothScanSkimmersCallback(), false);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp:2428:20: error: no matching function for call to 'NimBLEScan::start(int, void (&)(NimBLEScanResults), bool)'
 2428 |     pBLEScan->start(0, scanCompleteCB, false);
      |     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Rodrigo\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:22,
                 from C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.h:12,
                 from C:\Users\Rodrigo\Desktop\ESP32Marauder-1.0.0\ESP32Marauder-1.0.0\esp32_marauder\WiFiScan.cpp:1:
C:\Users\Rodrigo\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEScan.h:65:25: note: candidate: 'bool NimBLEScan::start(uint32_t, bool)'
   65 |     bool                start(uint32_t duration, bool is_continue = false);
      |                         ^~~~~
C:\Users\Rodrigo\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEScan.h:65:25: note:   candidate expects 2 arguments, 3 provided
Multiple libraries were found for "LinkedList.h"
  Used: C:\Users\Rodrigo\Documents\Arduino\libraries\LinkedList
  Not used: C:\Users\Rodrigo\Documents\Arduino\libraries\DisplayMenu
Multiple libraries were found for "SD.h"
  Used: C:\Users\Rodrigo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\SD
  Not used: C:\Users\Rodrigo\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: 'esp_base_mac_addr_set' was not declared in this scope; did you mean 'esp_base_mac_addr'?

FrancoSpermasolido, 2024-07-24:
did you find a solution yet? i'm having this issue as well

ffsg12345678, 2024-07-24:
I'm the same, I added extra files called "EspSoftwareSerial" and "MicroNMEA" in the Arduino IDE, otherwise it wouldn't compile

iceshadow2107, 2024-12-10:
> I'm the same, I added extra files called "EspSoftwareSerial" and "MicroNMEA" in the Arduino IDE, otherwise it wouldn't compile我也是一樣，我在Arduino IDE中新增了名為「EspSoftwareSerial」和「MicroNMEA」的額外文件，否則無法編譯

How can you added extra files called "EspSoftwareSerial" and "MicroNMEA" in the Arduino IDE?

avipars, 2024-12-25:
Does the ESP have it's own microsd card slot? See if you can go to configs.h and ensure that HAS_BT is commented out for your board (if it is there)

Inspirati, 2025-02-11:
simply add a forward declaration immediately before the function that calls it:

esp_err_t esp_base_mac_addr_set(uint8_t *Mac);

closed 2026-10-18 by github-actions[bot]