justcallmekoko/ESP32Marauder #559: I hope to increase support for the LILYGO T-Watch S3 ESP32-S3 Watch LoRa

YANGKUN87, 2024-08-02:
[image]

avipars, 2024-12-25:
Hey, which version of the esp32s3 is on your watch?

YANGKUN87, 2025-01-27:
@avipars Sir, my watch is ESP32S3, but I couldn't find any available firmware. Do you have  firmware?

avipars, 2025-01-27:
> [@avipars](https://github.com/avipars) Sir, my watch is ESP32S3, but I couldn't find any available firmware. Do you have firmware?

Unofficially there is, I was able to flash an esp32 s3 zero using the multiboard firmware in the github releases. Unfortunately, the bluetooth functionality will not work with this firmware. 

Additionally, as your device has a screen and buttons, if you could flash the firmware... you wouldn't be able to make much use of the lilygo hardware features.

It is best to fork this repo, and make the necessary changes and then compile for your device.

closed 2026-10-18 by github-actions[bot]