justcallmekoko/ESP32Marauder #564: Touch direction is inverted

SSnoody, 2024-08-16:
**Describe the bug**
Touch of the screen 2.4" got inverted

**Expected behavior**
The touch are inverted in all the menu's

https://github.com/user-attachments/assets/3ade7aeb-fb61-419f-9c0d-932b4f8e540b

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v1.0.0]

b00mekk, 2024-08-27:
try modifying all TFT SHIELD calData to:
uint16_t calData[5] = { 293, 3547, 243, 3592, 3 };

Fattcat, 2024-10-14:
Hello, I HAVE EXACTLY SAME PROBLEM !
Please HELP ME how can I Fix it ?

Fattcat, 2024-10-15:
HELLOO. I FIXED IT !!!
go to here : https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino
NEXT step is to Upload that code and calibrate it :D

NoIdeaWhatImDoing-yes, 2024-10-26:
> HELLOO. I FIXED IT !!!
> 
> go to here : https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino
> 
> NEXT step is to Upload that code and calibrate it :D
> 
> 
> 
> But ... Idk why IAM NOT ABLE TO COMPILE Arduino esp32 Marauder code from source when I edited ONLY TFT Display uint16_t calData[5] = { Here My Numbers };
> 
> 
> 
> error is here :
> 
> exit status 1
> 
> 
> 
> Compilation error: control reaches end of non-void function [-Werror=return-type]

I need help dude. I just started out and just built the esp32 marauder and I have the same problem, you told us how to fix it but I'm lost. Can you send me a vid or something, maybe even detailed instructions? If you do thank you so much!

Fattcat, 2024-10-29:
Hello Im finally here.
What is your problem ?
What u did and how (i need your steps to help u fix that)

Recently I started working on Shark Predator Firmware (more advanced abd hard now to do) this firmware will be able to do lot of types of attacks.

casoncure, 2024-11-03:
@Fattcat pls make a bin code too hard :(

Fattcat, 2024-11-03:
@casoncure 
Hello.
I want to know if u followed my step by step guide (if it was useful or was NOT)

YES, I can make **.bin**  file for you.
Pls wait is too early for me.

biba0807, 2024-12-08:
> @casoncure Привет. Я хочу знать, следовали ли вы моему пошаговому руководству (было ли оно полезным или НЕ было)
> 
> ДА, я могу сделать **.bin** файл для вас. Я СДЕЛАЛ ЭТО! Теперь вы можете скачать и прошить его отсюда
> 
> * (Просто) скачайте файл .rar, распакуйте его в папку, затем откройте esp32 flasher и выберите файлы .bin, которые вы распаковали.
> * Подпишитесь на это фото, если у вас нет абсолютно никакого опыта или навыков.
> * ИНФОРМАЦИЯ! Перепрошивка может занять 5-10 минут (зависит от вашего супербыстрого ПК xD) 
>   !
>   ![esp32-BIN-Flasher-руководство](https://private-user-images.githubusercontent.com/109801079/382589888-4f262825-6369-4d3e-832e-939ea07213d9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM2NTkwNzEsIm5iZiI6MTczMzY1ODc3MSwicGF0aCI6Ii8xMDk4MDEwNzkvMzgyNTg5ODg4LTRmMjYyODI1LTYzNjktNGQzZS04MzJlLTkzOWVhMDcyMTNkOS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOFQxMTUyNTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00YjFiOTA5ZThmNDkxNmRhNzhiZjIyOGUxZDgwZDgxMjFmMzczYTdhM2FmMDUyYmY2MDc3YTgwZjEwNmNjY2IwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sHrxZpwpZw28s94unVyfL6Xc9-G09l2u_m69NxCdtvc)
> * Дайте мне знать

where is rar itself?

Fattcat, 2024-12-08:
Rar is like file extention with saved files inside in 1 pack.
Search windows 10 rar download and install it.

biba0807, 2024-12-08:
> Rar это как расширение файла с сохраненными файлами внутри в 1 пакете. Найдите windows 10 rar скачайте и установите его.

I meant where to download the archive with this fix

Fattcat, 2024-12-08:
**go here** https://github.com/Fattcat/esp32-marauder (if is alreadi edited)

biba0807, 2024-12-08:
> **идите сюда ** https://github.com/Fattcat/esp32-marauder (он уже отредактирован)

[image]

ineqq, 2024-12-13:
> @casoncure Hello. I want to know if u followed my step by step guide (if it was useful or was NOT)
> 
> YES, I can make **.bin** file for you. I DID IT ! You can now download and flash it from here
> 
> * (Just) download .rar file, unpack it to folder, then open that esp32 flasher and pick .bin files that u unpacked
> * Follow this photo if u have absolutelly no experiences or skills
> * INFO ! it may take 5 - 10 mins to flash (depends on your super fast PC xD)
>   ![esp32-BIN-Flasher-tutorial](https://private-user-images.githubusercontent.com/109801079/382589888-4f262825-6369-4d3e-832e-939ea07213d9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxMjM2MDUsIm5iZiI6MTczNDEyMzMwNSwicGF0aCI6Ii8xMDk4MDEwNzkvMzgyNTg5ODg4LTRmMjYyODI1LTYzNjktNGQzZS04MzJlLTkzOWVhMDcyMTNkOS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxM1QyMDU1MDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMWZiZWZlZGI3NmFmMDY2YzM1ZTlmNWI3NDZlMzJhNTRkMmU5YjFhNzE2ZmFhMDI0MmY3YjQxYWQzYzhhYTI1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.QtG7G30sbo86lGM9dsF_bKrQ4Hm6CLPR5qpTfK1PvfQ)
>   !
> * Let me know

hi, could you make the 1.1 version for me? i beg youuu

casoncure, 2024-12-14:
Np I figured it out thx tho

Eifeow, 2025-01-26:
I have the same problem my touch is turned by 90 degre but I dont understand it anymore what should I do

ineqq, 2025-01-26:
> I have the same problem my touch is turned by 90 degre but I dont understand it anymore what should I do

just flash the v6 version, it should fix the problem

almol, 2025-02-09:
https://github.com/justcallmekoko/ESP32Marauder/issues/564#issuecomment-2614519507
This don't work for me: esp32_marauder_v1_2_1_20250207_v6

TFT 2.4 ILI9341

andysdesigns, 2025-02-23:
i think the touch reversed issue is on V4. i flashed V6 and all is good. touch working as it is.

Mafiadoener36, 2025-04-30:
v6 is compatible with wroom? why than he still builds old hardware?!? @ineqq @andysdesigns 

@justcallmekoko would you answer/clarify this? If we own diy wroom with ili9341 we can use v6 now? why thn u build still old_hardware?

closed 2026-10-18 by github-actions[bot]