justcallmekoko/ESP32Marauder #565: Support for CYD Screen ESP3248s035

Hard2Borrow, 2024-08-17:
With optimism, I'd picked up a 3.5" version of the CYD, intentions of having a larger display for the Marauder functions/ better readability- I attempted to utilize the flasher available from [Fr4nkFletcher](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commits?author=Fr4nkFletcher) 

-- to no avail. 

Anyone can point this fellow in the right direction to get the firmware running on the 3.5" screen?

Asmodeux2, 2024-09-17:
Там есть нюансы с библиотекой TFT_eSPI нужно подобрать в User_Settings.h нужный модуль а если нет добавить самостоятельно [но скорее всего там какой-нибудь ILI9394]

Fr4nkFletcher, 2025-01-24:
@Hard2Borrow try now

closed 2026-10-18 by github-actions[bot]