justcallmekoko/ESP32Marauder #566: SD Card Issues

tcphdr, 2024-08-22:
**SD Card Mounting Issues**
I still cannot mount an SD card to my dev board no matter what card type I use. I've bought a gigastone 32gb from amazon, a sandisk 32gb from walmart. Unable to mount SD card to marauder when used with flipper zero.

**To Reproduce**
Steps to reproduce the behavior:
1.) Mount wifi dev board to flipper zero
2.) turn on flipper
3.) check boot log for wifi devboard.

**Expected behavior**
The SD card should mount from the FZ.

**Screenshots**
N/A

**Marauder (please complete the following information if applicable):**
- Latest Marauder
- RogueMaster Latest Firmware on FZ

**Desktop (please complete the following information if applicable):**
 - OS: Win11
 - Browser: Opera GX

Not sure what my issue is, or why I keep encountering this issue. PLEASE NOTE: all cards are formatted in FAT32, even out-of-the-box gigastone cards refused to work. I am NOT using any modifications aside from a flipper zero and the wifi dev board, which should be all that I need.

zangoborillo, 2024-09-18:
I also have the same problem, after making last update " No Core Dump partition found!,Unable 
mount the SD card

avipars, 2024-12-25:
With the official dev board from Flipper Zero?

tcphdr, 2025-01-05:
Yes.

closed 2026-10-18 by github-actions[bot]