justcallmekoko/ESP32Marauder #572: RF signal

neyzenu, 2024-09-02:
Marauder is perfect but there should be a place for RF in it too
It would be truly incredible if many features were added such as copying RF signals, recording and replaying, monitor mode, jamming start etc
and 
I believe that if infrared is added, the project will reach a truly incredible level.
like flipperzero
thank you in advance

SuthaharanTharunjan, 2024-09-03:
I also think that if esp32 has enough pins and space for additional code, it is possible to implement other flipperzero like features 
NFC/RFID ,Sub ghz,IR
**Scanning copying and emulating, I mostly like those features because it is practical to use like a multitool **

Fattcat, 2024-10-14:
Yes definitelly !
Can some RE-Make it and add that stuff like subghz, RF, CC1101 (keyfob clonning, replay attack) ?
Thank you & have a nice day.

YANGKUN87, 2025-01-27:
T-Embed CC1101

Fattcat, 2025-01-27:
But how with T embed ?

closed 2026-10-18 by github-actions[bot]