justcallmekoko/ESP32Marauder #578: White screen Help Fast if possible

esp32-ceeling, 2024-09-19:
**Describe the bug**
White screen but cli works Serial monitor shows no issues i tried everything if theres something that i cloud do please comment below or contact me on discord dc : awp13

**To Reproduce**
I dont know how to reproduce it yet.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Sorry i dont have screenshots here but i have some on my discord

**Marauder
 - Firmware version: [e.g. v1.0.00]
 - Hardware version: [e.g. OG] 

**Desktop 
 - OS: [Windows 11]
 - Browser [Chrome]
 - Version [Latest]

Thinkhourse, 2024-10-14:
same problems

esp32-ceeling, 2024-10-26:
bought a cyd and works

yusa006, 2025-01-25:
> bought a cyd and works

what is that bro

JLowe269, 2025-01-28:
Same issue here. V4