justcallmekoko/ESP32Marauder #580: BT Wardrive continuous infinite loop

carson-evans, 2024-09-22:
**Describe the bug**
BT Wardrive continuous infinite loop on awok mini v2. Clicking button does not exit, can only reboot.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Bluetooth > Sniffers > BT Wardrive Continuous
2. Click on the function
3. Infinite loop begins
4. Clicking interactive button does not exit function, must reboot

**Expected behavior**
Clicking button on awok mini v2 should exit function

**Marauder (please complete the following information if applicable):**
 - Firmware version: v1.0.0
 - Hardware version: Marauder Mi (The screen cuts off the rest)

**Additional context**
Add any other context about the problem here.

carson-evans, 2024-09-23:
Pull Request #581 Missing close bracket.

justcallmekoko, 2025-03-26:
I am unable to reproduce this issue on any of my mini hardware.

closed 2026-10-18 by github-actions[bot]