justcallmekoko/ESP32Marauder #586: White screen on TFT display

Mantas22XYT, 2024-10-06:
**Describe the bug**
The 2.8 TFT Screen is all white when booted.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the screen to the TFT using provided IC connections
2. Connect ESP32 to power
3. White screen

**Expected behavior**
I expect the Marauder menu to be displayed on bootup.

**Screenshots**
[image]

**Marauder (please complete the following information if applicable):**
 - Firmware version: v1.0.0
 - Hardware version: NodeMCU LDDB

Mantas22XYT, 2024-10-06:
Nevermind I'm a fucking idiot

closed 2024-10-06 by Mantas22XYT

pepino289, 2025-01-08:
what was the problem pls explain

Mantas22XYT, 2025-01-08:
check if your pins are defined correctly in the display's setup.h file