justcallmekoko/ESP32Marauder #588: v1.0.0 doesn't compile

Mantas22XYT, 2024-10-07:
**Describe the bug**
Marauder v1.0.0 doesnt seem to compile at all without having to modify the WifiScan and LedInterface files in some way. v0.13.10 compiles just fine without any modifications, but the screen is all white on bootup, becomes grey after a few seconds and then goes straight to white again. How could I fix this?

**To Reproduce**
Steps to reproduce the behavior:
1. Try to compile v1.0.0
2. Fuck ton of errors, mostly related to `MEM_LOWER_LIM` not being declared in `WiFiScan::checkMem()` and `LedInterface::RunSetup()` trying to run despite not being enabled.

**Expected behavior**
The project should compile, yet it doesn't.

**Marauder (please complete the following information if applicable):**
 - Firmware version: v1.0.0
 - Hardware version: GENERIC_ESP32

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Browser: Firefox
 - Version: 131

**Additional context**
Related: #585

Fattcat, 2024-10-22:
Can u be MORE speciffic ?

Mantas22XYT, 2024-10-22:
It's impossible to compile the project without encountering errors. The last version that compiles normally is 0.13.10.

Fattcat, 2024-10-23:
Ahh im so disapointed.

I tried compile v 0.13.10 and nothing work at all :(

avipars, 2024-12-25:
I spent time making changes and was able to get the build to compile on a esp32s3 mini. Basically if your ESP or board isn't on his officially supported list, good luck getting it to work.

sgstudio2024, 2025-05-02:
已经绝望了（）

closed 2026-10-18 by github-actions[bot]