justcallmekoko/ESP32Marauder #591: White Screen after upload firmware

risorsa, 2024-10-14:
when I upload the firmware to the Spacehuhn website following the instructions correctly and as asked I do RESET, the screen remains white 
i have: tft LCD 2.4 ILI9341 and ESP32-WROOM-32 NodeMCU, driver CP2102

(I tried a base code to see if the screen still remained white but the screen worked correctly)
[image]

Mantas22XYT, 2024-10-15:
You should check your `User_Setup.h` file in the TFT_eSPI library's installation folder. Scroll down to the pin definitions and change the GPIO pins to the ones listed in the IC connections page of the wiki.

risorsa, 2024-10-15:
i have Installed Firmware via OTA

risorsa, 2024-10-15:
> You should check your `User_Setup.h` file in the TFT_eSPI library's installation folder. Scroll down to the pin definitions and change the GPIO pins to the ones listed in the IC connections page of the wiki.

now when I install the firmware the screen no longer remains white but this happens and sometimes you see the initial loading writings but then it remains in this state
[image]

sapmejinjer, 2024-10-20:
you might be using st7789 instead of ili9341, i also have that issue but then i switched displays and it works

sapmejinjer, 2024-10-20:
> you might be using st7789 instead of ili9341, i also have that issue but then i switched displays and it works
if its not the display, its either the flashing issue

closed 2026-10-18 by github-actions[bot]