justcallmekoko/ESP32Marauder #595: Marauder esp32 honeyhoneytrading from etsy not booting up again after firmware update via sd card.

S3real-ux, 2024-10-28:
**Describe the bug**
I got the marauder from honeyhoneytrading on Etsy 3 weeks ago, got it today, worked fine played around with it until I saw I needed to update the firmware, went on here, found the latest firmware update, downloaded the latest .bin file put it on my SD card and renamed it to update.bin and put it back in my marauder, went to device, firmware update, SD card update, yes, update successful and said rebooting and never rebooted, so I powered it off manually and it never booted up again, all the lights go on but the screen stays black. 

**To Reproduce**
Steps to reproduce the behavior:
1. SD card update
2. Click on update using SD card, click yes.
3. says update success full, rebooting and never reboots up again and manually turn it off and on again
4. device doesn't boot up again, black screen.

**Expected behavior**
to reboot automatically and no black screen and to work normally again.

https://github.com/user-attachments/assets/589ec685-0fe9-46af-8477-15995ab9131c

**Screenshots**
https://github.com/user-attachments/assets/2d17f73d-273e-4b81-b1c9-65292d47e2b8
https://github.com/user-attachments/assets/f8104696-a507-48a0-aa0f-0ba83cc53130
https://github.com/user-attachments/assets/589ec685-0fe9-46af-8477-15995ab9131c

**Marauder (please complete the following information if applicable):**
 - Firmware version: esp32_marauder_v1_0_0_20240626_esp32_lddb.bin
 - Hardware version: got it newly and didn't check so I don't know sadly 

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Browser: OperaGX
 - Version: uhm the latest version

**Additional context**
the .bin file bricked my working marauder, no idea how to fix it now. really disappointed

justcallmekoko, 2024-10-28:
Because this is not officially supported hardware, I don't know what bin file you're supposed to use. I know for a fact you are not supposed to use the LDDB bin file though as this is meant for the LDDB attachment to the Flipper Zero which does not have a display. Your unit is still probably running the Marauder firmware and is even available over command line via serial terminal. Unfortunately because it is now running the LDDB version of the firmware, you will no longer be able to use the SD update option as the firmware is configured to use the SD card pins for the LDDB hardware...which is not what you have. You will need to use the Web Updater method of reinstalling the firmware. But you will need to contact the original manufacturer of that device to find out what version of the bin file they use to flash their devices. Hopefully they offer support.

closed 2024-10-28 by justcallmekoko

S3real-ux, 2024-10-28:
> Reply to this email directly, view it on GitHub
> You are receiving this because you authored the thread.Message ID:

Thanks a bunch for reaching back to me, I've contacted them and he
said I have to go get a esp32 debugger and he will help me Debug it.
Hopefully it will work because I have no experience at all into this

S3real-ux, 2024-10-28:
Thanks a bunch for reaching back to me, I've contacted them and he
said I have to go get a esp32 debugger and he will help me Debug it.
Hopefully it will work because I have no experience at all into this,
also I tried using web but it couldn't detect the marauder so it's
pretty bricked.

justcallmekoko, 2024-10-28:
Frankly it sounds like they are over complicating it. Just ask them what bin file they got off of my page so you can use the web installer. We have documentation in our wiki which will show you how to do it...but they need to tell you which file.

If they can't help you without you buying something or getting some random tool, let me know. I'll send you a real Marauder to replace that etsy knock off.

justcallmekoko, 2024-10-28:
And really we can trade. Send me the etsy knock off and I will send you a real v6. If I can fix the etsy knock off, I will send it back to you and you can just keep both

S3real-ux, 2024-10-28:
Would you really do that?, I mean tbh I payed 300swiss franks for it
with all the shipping costs and the taxes, but I thought
honeyhoneytrading is legit?.

S3real-ux, 2024-10-29:
For sure, I will send the links to you, is it possible to contact you
via discord? I'm not so good on chatting through email lol. This is my
user s3realest, and this is the link to what I bought:
https://www.etsy.com/listing/1684827211/esp32-marauder-pocket-unit-with-gps-v2.
I will get back to you tomorrow since it's midnight for me right now.
Thanks a bunch again it means a lot

S3real-ux, 2024-10-29:
Hey koko, I've texted you on discord. I also sent U the details of the
Etsy I bought the marauder from, any news?

S3real-ux, 2024-10-29:
Hello koko, I really help right now, he's making me pay more for the
postage to send the debugger, because I have tried using web but it
doesn't detect my marauder. Please help