justcallmekoko/ESP32Marauder #598: Errors with a DIY marauder

Zynth9999, 2024-10-31:
**Describe the bug**
Keeps triggering brownout on all cables i have and sends core dump errors

**Expected behavior**
Boot normally

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v1.0.0]
 - Hardware version: [e.g. v4] 

**Desktop (please complete the following information if applicable):**
 - OS: [Windows]
 - Browser [chrome]

Zynth9999, 2024-10-31:
```E (532) esp_core_dump_flash: N���ɕ dump partition found!

E (532) esp_core_dump_flash: No core dump partition found!

ESP-IDF version is: v4.4.5

Brownout detector was triggered

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1100

ho 0 tail 12 room 4

load:0x40078000,len:12308

load:0x40080400,len:3076

entry 0x400805ec

E (532) esp_core_dump_flash: N���ɕ dump partition found!

E (532) esp_core_dump_flash: No core dump partition found!

ESP-IDF version is: v4.4.5

Brownout detector was triggered

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1100

ho 0 tail 12 room 4

load:0x40078000,len:12308

load:0x40080400,len:3076

entry 0x400805ec

E (532) esp_core_dump_flash: N��ɕ dump partition found!

E (532) esp_core_dump_flash: No core dump partition found!

ESP-IDF version is: v4.4.5```

sapmejinjer, 2024-11-09:
probably unstable power

justcallmekoko, 2024-11-17:
This will be due to poor power supply either with the battery itself or the LDO. This is not a limitation of the firmware.

closed 2024-11-17 by justcallmekoko