justcallmekoko/ESP32Marauder #600: Bluetooth GPS support.

derGreen, 2024-11-05:
**No solder GPS receivers. Use bluetooth** 

Ther are Portable BT GPS receiver out there, was used more frequently a few years ago. 
Before smartphones were widespread. Used ones are cheap to get 

Since the ESP32 supports Bluetooth, it would be good in this case for wardriving and collecting Wifi networks if you could connect a GPS receiver via Bluetooth. Such devices can be obtained cheaply second-hand. No soldering to the hardware is necessary. External Bluetooth receivers send the current coordinates via serial too, so it should be as easy to integrate in the code as wired receivers.

justcallmekoko, 2024-11-17:
You cannot use bluetooth and wifi functions at the same time. It is a hardware limitation of the esp32

derGreen, 2024-11-17:
Oh, damn :(