justcallmekoko/ESP32Marauder #615: St7789 display landscape to portrait

Keepoveon, 2024-11-26:
**Describe the bug**
St7789 display landscape 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v0.9.17]
 - Hardware version: [e.g. OG] 

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information if applicable):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hothien614, 2024-12-10:
how to fix it please guide me.

justcallmekoko, 2025-05-30:
https://github.com/justcallmekoko/ESP32Marauder/pull/776

closed 2025-05-30 by justcallmekoko