justcallmekoko/ESP32Marauder #626: Show the Frequency of each WIFI Network in the AP list (2.4, 5)

Killer74-hub, 2024-12-19:
**Is your feature request related to a problem? Please describe.**
I am frustrated, when I don´t realize that I am trying to deauther a 5ghz Network. That´s why, Listing the Strength, would be nice.

**Describe the solution you'd like**
List the Frequencys in the "List" Tab.

**Describe alternatives you've considered**
Checking it via an other device/Programming an script for it. But it takes too long.

**Additional context**
Nope

ToxxikHalo, 2024-12-19:
Marauders of any esp32 varient canton lyrics scan for and list the 2 4Gzhz networks... if it's a ual band network it'll only see the 2.4GHz side...

So what's listed is all 2 4GHz. So it'd be redundant at that point u know...n

Killer74-hub, 2024-12-20:
> Marauders of any esp32 varient canton lyrics scan for and list the 2 4Gzhz networks... if it's a ual band network it'll only see the 2.4GHz side...
> 
> So what's listed is all 2 4GHz. So it'd be redundant at that point u know...n

Uhm so, the thing is, some flipper zero wifi development boards will have 5ghz and mine has too I think. It has an antenna on it and on the board it said 2.4-5ghz

justcallmekoko, 2025-04-01:
ESP32 currently does not support 5ghz

closed 2025-04-01 by justcallmekoko