justcallmekoko/ESP32Marauder #630: Support for TFT 2.8 inch ST7789 display?

avdol0521, 2024-12-26:
when flashing with the firmware provided. the TFT 2.8 inch ST7789 display im using displays everything in landscape mode but the actual touch input acts as if it was in portrait so its next to impossible to use properly. i cant even select an AP because the UI gets squished and fuzzy when i try. its not a problem with my display i tried flashing with a fork made by DonAnonymousIo and it works just fine but its a reaaallllyy old version of the firmware and is missing alot of features. it would be a big help if you could add support for the display or at the very least give me a description of how i could port it for my display myself. heres an image of what the issue looks like
[image]

PGF123, 2025-01-06:
https://github.com/clipboard1/ESP32Marauder-ST7789/releases/tag/v0.10.4

Roughneckdalarna, 2025-01-16:
i had the same problem -- and used this firmware  and diplay is just white .. plz help

justcallmekoko, 2025-05-30:
https://github.com/justcallmekoko/ESP32Marauder/pull/776