justcallmekoko/ESP32Marauder #639: ble issue

Blaza-4, 2025-01-14:
i am trying build the firmware on Arduino ide . i downloaded all the libraries but when i complie this error: no matching function for call to 'NimBLEAdvertisementData::addData(std::__cxx11::string)' comes always. is something wrong in the code or library i am using . i used the zip file provided on this repo

Fr4nkFletcher, 2025-01-20:
Are you using NimBLE 1.3.5?

RaceNJason, 2025-01-28:
Not sure why this is still using v1.3.5 when v2.2.x is out...but, I'm sure there is a reason. To help with the build I went through all the libraries to document which versions the build was using (and then saved it to the readme.txt file in the libraries folder of the marauder project). This was helpful since I had to roll back a few and make sure not to select the 'update all' when the Arduino IDE asks if I want to update my libraries. This was core to my problem since I've always had the habit of using the latest versions and updating my code if needed. Maybe this might help someone else...hope my formatting holds when this posts.

Library			Expected Version		Latest		Author
Adafruit_BusIO			1.15.0		1.17.0		Adafruit
Adafruit_GFX_Library		1.11.9		1.11.11		Adafruit
Adafruit_ILI9341			1.6.0			1.6.1			Adafruit
Adafruit_MAX1704X		1.0.3			1.0.3			Adafruit
Adafruit_NeoPixel			1.12.0		1.12.4		Adafruit
Adafruit_SH110X			2.1.10		2.1.11		Adafruit
Adafruit_STMPE610			1.1.6			1.1.6			Adafruit
Adafruit_TouchScreen		1.1.5			1.1.5			Adafruit
Adafruit_TSC2007			1.1.2			1.1.2			Adafruit
ArduinoJson				6.21.4		7.3.0			Benoit Blanchon
AsyncTCP					1.1.4			1.1.4			dvarrel
ESP32Time				2.0.4			2.0.6			fbiego
ESPAsyncTCP				1.2.4			1.2.4			dvarrel
ESPAsyncWebServer		2.10.4		3.6.0			Me-No-Dev (Mathieu Carbou)
EspSoftwareSerial			8.1.0			8.1.0			Dirk Kaar, Peter Lerup
ILI9341_t3-master			1.0						Limor Fried, Adafruit, Paul Stoffregen
       Used the below library instead (proper name in Arduino libraries list)
	ILI9341_t3n			1.0			1.1.1			Limor Fried, Adafruit, Paul Stoffregen
JPEGDecoder				2.0.0			2.0.0			Bodmer
LinkedList				1.3.3			1.3.3			Ivan Seidel
LittleFS_esp32				1.0.6			1.0.6			lorol
LittleFS_Mbed_RP2040		1.1.0			1.1.0			Khoi Hoang
LovyanGFX				1.1.9			1.2.0			lovyan03
lv_arduino				3.0.1			3.0.1			Gabor Kiss-Vamosi
	LVGL_CYD? have had a few errors with the above library and wondering if this one would be a better replacement
MicroNMEA				2.0.6			2.0.6			Steve Marple
NimBLE-Arduino			1.3.5			2.2.1			h2zero
NTPClient				3.2.1			3.2.1			Fabrice Weinberg
SwitchLib					1.1.1			1.1.1			justcallmekoko
TFT_eSPI-master			2.5.43					Bodmer
       Used the below library instead (proper name in Arduino libraries list)
	TFT_eSPI				2.5.43		2.5.43		Bodmer
ThingPulse_XPT2046_Touch	1.4		1.4		?		ThingPulse

justcallmekoko, 2025-03-25:
We are on NimBLE 2.2.3. Make sure you have this library installed when building

closed 2025-03-25 by justcallmekoko

CodeHedge, 2025-06-02:
I had the same error during compile. I had to downgrade to 1.4 and it worked after that.

Not sure why :/

hossammorayan, 2025-09-29:
am install all same library @RaceNJason tell it and install version you tell @justcallmekoko about NimBLE-Arduino, 1.3.5, 2.2.1, h2zero am install version 1.3.5  @RaceNJason and get error on it and try 2.2.3 same thing get error on , all error am get it it about version of nimble and error is different  in versions am try it  mean error am get by version 1.3.5 is different in 2.2.3 

what best version of esp32 espriesef system use because version 2.0.10 is get accses denied and compile am try in version 2.0.17

hossammorayan, 2025-09-29:
thank you for your reply I was hopeless that this would be answered in my problem

your imagine for am try to install last version thats right am try it and am solve some problem of nimble by work in last version of board manger (ESP32) (3.3.1) and other library by last version but after some try to solve am get some big errors  after it I was disappointed but i will try again and work with bin files at the moment 

your reply in paragraph 2 is right and am agreed it

when am try to compile by version of board manger (ESP32) (v2.0.10 Tell to use in wiki) , 2.0.14 ,2.0.13 some of it is install but other not install and get error 
Failed to install platform: 'esp32:esp32:2.0.14'.
Error: 2 UNKNOWN: stream error: stream ID 1; PROTOCOL_ERROR; received from peer

v2.0.10 get access denied in output terminal in arduino ide 
2.0.14 ,2.0.13 not need to install and get error above

why am choose this version am choose from youtube video and from wiki topic how to step Arduino ide but is still not work