justcallmekoko/ESP32Marauder #645: 5ghz Maruder

MimoAW, 2025-01-21:
Hi! I know this is asked alot but I was curious if 5ghz capabilities will ever come to maruder there are esp32s with 5ghz now and there are a few other projects that's cancelled so it but not as good as maruder like this: https://github.com/tesa-klebeband/RTL8720dn-Deauther

Thank you!

MimoAW, 2025-01-21:
Also congratulations on maruder v7!

Nigdzie, 2025-02-09:
we need to wait for https://www.espressif.com/en/products/socs/esp32-c5
this chip will change the game ;-)

MimoAW, 2025-02-09:
> we need to wait for https://www.espressif.com/en/products/socs/esp32-c5
> this chip will change the game ;-)

I honestly thought it was already out 🤣

Nigdzie, 2025-02-10:
heh no it was just a news, now maybe after 3-4 weeks first orders could be placed.