justcallmekoko/ESP32Marauder #646: ILI9341 TFT Display Not Working with ESP32 - Stuck on White Screen

yusa006, 2025-01-25:
Hi everyone,

I’m currently working on a project using a 2.4" TFT SPI 240x320 ILI9341 display and an ESP-Wroom-32. I’ve connected the display to the ESP32 using jumper wires on a breadboard, following the pinout and wiring as described in the documentation. However, I’m facing an issue, and I hope someone can help me out.

When I power up the circuit:
The screen initially turns dark gray for 2-3 seconds.
After that, it turns completely white and stays that way.
The expected Marauder application does not appear on the screen.

[image]

What I’ve Done:
Double-checked all the connections to ensure they match the pinout for the ILI9341 and ESP32.
Verified that the ESP32 is receiving power and the code is uploaded correctly.
The display is powered via the ESP32’s 3.3V pin.

[image]

Could anyone help me figure out why the display is not working as expected? Any advice or suggestions would be greatly appreciated!

Thanks in advance for your help!

Fr4nkFletcher, 2025-01-25:
Get a TFT-eSPI example sketch to work first and I assure Marauder will follow

divine-interventi0n, 2025-02-14:
[image]

[image]

Triple check the pinout, hope this helps! If you’re building with recommended hardware for the original DIY guide for v4 Marauder, make sure you’ve followed the proper steps when flashing the device.  (Both “README.md” and “old_README.md” should be used for v4 hardware)

blazedmaster-spec, 2026-08-29:
@divine-interventi0n i know your setting says busy but can you please tell me step by step how to install the compatible firmware for the ESP32 WROOM 32D (thats the newer version of the regular ESP 32) and my screen is also white and Im not sure if i have the correct wiring complex. So if you have the time please reply