justcallmekoko/ESP32Marauder #648: request: add support for ST7796 TFT driver

HoaxParagon, 2025-01-30:
The driver is found on the Lilygo T-Display S3 Pro.

Fr4nkFletcher, 2025-01-31:
Check out this [commit](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/58b1002f01c68f151b2fc598a45f16585419a09d#diff-c4abe84c95dad848e5037e170b9db85b9d56ad913043fb7ea1b80618348c25deR438) 

Fairly straightforward if you're compiling from source.

HoaxParagon, 2025-02-08:
> Check out this [commit](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/58b1002f01c68f151b2fc598a45f16585419a09d#diff-c4abe84c95dad848e5037e170b9db85b9d56ad913043fb7ea1b80618348c25deR438)
> 
> Fairly straightforward if you're compiling from source.

what would I do for the user_setup.h file?

closed 2026-10-18 by github-actions[bot]