justcallmekoko/ESP32Marauder #667: Blank / white screen after boot using WROOM32E

pepspeps-hack, 2025-02-15:
After flashing my NodeMCU WROOM32E v1.3 used in combination with the 2.8 TFT SPI 240x320 v1.2 touchscreen I only get a white screen. I tried flashing using the ESPwebtool, FZEE Flasher and using the commandline flashtool  FZ_Marauder_v2.8. I checked and rechecked the  readme and the old readme for the correct pinout. 

I reproduced this issue with another ESP32 board using the WROOM32 on a Devkit V1 board. 

I used the following commandlines with the Windows Flashtool v 2.8:

To erase:

> esptool.exe -p COM4 -b 921600 -c esp32 --before default_reset -a no_reset erase_region 0x9000 0x6000

Ouput:
```
esptool.py v4.1
Serial port COM4
Connecting.......
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 14:2b:2f:db:ea:d4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.1 seconds.
Staying in bootloader.
```

Then flashing the firmware with the following command:

> esptool.exe -p COM4 -b 921600 --before default_reset -a hard_reset -c esp32 write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 WROOM\bootloader.bin 0x8000 WROOM\partitions.bin 0x10000 WROOM\esp32_marauder_v1_0_0_20240626_old_hardware.bin 0xE000 WROOM\boot_app.bin

I get the following output:

```
esptool.py v4.1
Serial port COM4
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 14:2b:2f:db:ea:d4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00182fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Warning: Image file at 0x1000 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=2MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 16576 bytes to 11405...
Wrote 16576 bytes (11405 compressed) at 0x00001000 in 0.4 seconds (effective 322.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 430.1 kbit/s)...
Hash of data verified.
Compressed 1515632 bytes to 951067...
Wrote 1515632 bytes (951067 compressed) at 0x00010000 in 15.5 seconds (effective 784.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 976.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

After that I only get a blank / white screen. What am I doing wrong? Please some advice...! Is additional info needed?

Lorrpozzz, 2025-02-15:
got the same problem please help

Pommes20304050, 2025-03-28:
Same here ^^ we need to flash it in a different way i guess, i think you both are here because of the YT Video where he flashed it via ESPWebTool...

Pommes20304050, 2025-03-28:
# Solution for both of you (@Lorrpozzz, @pepspeps-hack)

Here’s the solution:

Go to the following website:  
https://fr4nkfletcher.github.io/Adafruit_WebSerial_ESPTool/  

Choose your board.  
In my case, it was the regular "cheap yellow 2.8-inch display without GPS with the USB-C port".  

Good luck!