justcallmekoko/ESP32Marauder #692: Flipper Zero - 1.4.x - Fail to set STA MAC

Nigdzie, 2025-04-07:
**Describe the bug**
Since 1.4.0 all deauth attack end with error"

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Scan AP
2. Select AP
3. Attack Deauth

**Expected behavior**
On old 1.2.1 attack is working fine without any errors.

**Screenshots**

[image]

[image]

**Marauder (please complete the following information if applicable):**
 - Firmware version: [e.g. v1.4.0 / 1.4.1]
 - Hardware version: [e.g. Flipper Zero Wifi Module V1 + Flipper Zero SD-GPS mod]

Phaneron1991, 2025-04-07:
Had the same issue when I flashed 1.4.0 last night, I'm running mntm-009
On flipper dev board and LDDB both updated with update.bin on sd

Brasshopper, 2025-04-09:
Same issue here, is the old version available somewhere?  I have exactly this issue with a Flipper dev board.  I used FZEE Flasher to install the Marauder code on the Flipper WiFi Developer board.  I was unsure as to whether the message mattered, I'm new to Flipper.

Nigdzie, 2025-04-09:
> Same issue here, is the old version available somewhere? I have exactly this issue with a Flipper dev board. I used FZEE Flasher to install the Marauder code on the Flipper WiFi Developer board. I was unsure as to whether the message mattered, I'm new to Flipper.

Hey,
yes here you go, latest working version:
https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v1.3.0

lewekleonek, 2025-04-09:
I don't see ESP_OK assigned or defined anywhere (yes, I've search the whole code repo)... but I could've missed something, because I can't delve into it right now:

https://github.com/justcallmekoko/ESP32Marauder/blob/f0d1c048f5f542eb77fa7dfcea0d7382b3ea614d/esp32_marauder/WiFiScan.cpp#L1722

justcallmekoko, 2025-04-09:
This is normal behavior from the Marauder firmware. The reason you see this error message is because AP and Station MAC address may only be set when the respective interface is initialized. i.e. if you go to set the AP MAC address, the AP interface must be initialized. If you go to set the Station MAC address, the Station interface must be initialized.

If you are using a function which does not require the Station interface, the Station interface will not be initialized. This is done in order to save memory. Because of this, you will see an error message saying the station MAC address failed to initialized. Even though you see this error, you should note that the function you are trying to execute still executes normally. We tested this to ensure proper functionality.

You can resume using v1.4.0 as normal. If you notice specific functionality not working as intended, please submit a new issue.