justcallmekoko/ESP32Marauder #703: ESP32 Select HTML Not Working

murph146, 2025-04-10:
**Describe the bug**
When selecting an .html file from Wifi > General > Select EP HTML file and navigating back to Evil Portal the selected html file is not selected/display in the Evil Portal site or displayed on the LCD screen

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Wifi > General > Select EP HTML
2. Click on 'listed html file'
3. Click on "back"
4. Navigate back to Wifi > Attacks > Evil Portal

**Expected behavior**
Expect the selected html to be displayed in the Evil Portal

**Marauder (please complete the following information if applicable):**
 - Firmware version: 1.4.2
 - Hardware version: Marauder 6.1

justcallmekoko, 2025-04-12:
Do you mean the captive portal does not display the proper web form in the browser or do you mean the name of the html file is not displayed on the Marauder screen?

I am unable to reproduce any of these errors

murph146, 2025-04-17:
Sorry for the delayed response. If it helps I can share a video. Basically, if i select AT&T.html from the list when i startup Evil Portal it's still displaying Spectrum.html file web form..

All files are in the root of the SD card and including the ap.config.txt file.

murph146, 2025-04-17:
I reformatted the card and copied all the files over and magically it seems to be loading the html file I'm selecting. Very weird...

The only thing I noticed is the Evil Portal seems to occasionally just crash and reboot the device haven't been able to tie to any specific action though.