justcallmekoko/ESP32Marauder #704: Add functionality to EvilPortal.cpp to grab selected wifi ap name for dynamic html portal display

1cyb3rpunk, 2025-04-10:
Is your feature request related to a problem? Please describe. 

Yes, the current version of the ESP32 Marauder's Evil Portal lacks the ability to dynamically update the display with the network's SSID (access point name). This limitation means that the captive portal always displays static information, such as "Netgear," which might not align with the actual network name being hosted. I'm frustrated because this lack of dynamic functionality reduces the immersive and customized user experience for the captive portal.

Describe the solution you'd like I would like the Evil Portal to dynamically fetch the SSID of the network being hosted by the ESP32 and update the displayed information accordingly. This update should:

Replace all references to the generic "Netgear" name with the actual network name (e.g., "ESP32_Marauder" or another configured SSID).

Ensure this dynamic functionality works seamlessly with the existing features of the captive portal, such as IP address fetching and form submissions.

Describe alternatives you've considered An alternative solution could involve manually updating the HTML file each time the network name changes. However, this is impractical, especially in environments where the ESP32 hosts multiple networks or uses variable SSIDs. Another approach could involve using hardcoded SSIDs in the HTML file, but this would require recompiling the code every time the access point name is modified.

Additional context for EvilPortal.cppp

fetch('/get-network-name')
    .then(response => response.text())
    .then(networkName => {
        document.getElementById('network-name').innerText = `${networkName} Connection Update`;
        document.getElementById('footer-name').innerText = networkName;
    })
    .catch(error => {
        console.error('Error fetching network name:', error);
    });

and 

server.on("/get-network-name", HTTP_GET, [](AsyncWebServerRequest *request) {
    String networkName = WiFi.softAPSSID(); // Fetch the SSID dynamically
    request->send(200, "text/plain", networkName);
});

justcallmekoko, 2025-04-12:
I am not sure exactly what you are asking. Can you please provide screenshots and highlight the issue you are talking about? Currently you can start an evil portal session with a selected SSID. The selected SSID will be displayed on the Marauder screen to indicate which SSID is being used. The SSID will then appear in client devices under the list of available WiFi access points. When the evil portal access point is selected by a client device and the captive portal is served, the access point name is served at the top of the captive portal banner.

If the evil portal session is ended, another SSID selected, and evil portal restarted, the new SSID is reflected.

justcallmekoko, 2025-04-12:
[image]

1cyb3rpunk, 2025-04-12:
You are correct the ssid will be presented on the screen for the captive portal, however it cannot be used with in the content of the html page dynamically. If I create an index.html and customize it I am asking to expose the ssid as a parameter so I can use it dynamically in the page via script. This will allow for better customization of the index.html page.

Sample Index.tml
<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <title>Dynamic SSID and Copyright</title>
</head>
<body>
    <h1>Welcome to the Captive Portal</h1>
    <p>Connected to: <span id="network-name">...</span></p>
    <p>You have connected to the <span id="network-name"> network. Please accept terms to connect.</p>
    <footer>
        <p>© 2024 <span id="footer-name">Copyright. </span>. All rights reserved.</p>
    </footer>

    <script>
        // Fetch the SSID dynamically from the server
        fetch('/get-network-name')
            .then(response => response.text())
            .then(networkName => {
                // Update the SSID in the main text and copyright footer
                document.getElementById('network-name').innerText = networkName;
                document.getElementById('footer-name').innerText = networkName;
            })
            .catch(error => {
                console.error('Error fetching network name:', error);
                document.getElementById('network-name').innerText = 'Error fetching SSID';
                document.getElementById('footer-name').innerText = 'Network Name';
            });
    </script>
</body>
</html>

1cyb3rpunk, 2025-04-16:
Any change of getting this in the next firmware release?

closed 2025-04-30 by 1cyb3rpunk