justcallmekoko/ESP32Marauder #705: CYD ESP32-2432S032C

wertsaco, 2025-04-11:
I would love it if there were compatibility with the CYD ESP32-2432S032C device. Currently, it only has support for an old version of Marauder, and I’m sure that since it’s one of the devices that has dropped significantly in price, it will become very common.

In advance, I appreciate this valuable community and the developer’s work.

zeuzeisloose, 2025-04-11:
There is a project someone else runs, they currently have it running version 1.3.1 of Marauder, as they usually lag behind a bit. Look in to the "CYM flasher", as I believe the model you are asking about is covered there. They are just porting Marauder to the CYD devices, so its the same software, but they are handling the touch drivers, etc.

wertsaco, 2025-04-11:
thnx a lot, I´ll keep an eye, thnx

closed 2025-04-11 by wertsaco

Fr4nkFletcher, 2025-04-13:
@wertsaco check [here](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/releases/tag/v1.4.2)

wertsaco, 2025-04-14:
@Fr4nkFletcher awesome thanks a lot!!!!