justcallmekoko/ESP32Marauder #706: ESP32-C3 SuperMini

Aboalfadl, 2025-04-13:
I tried to flash this board with the latest ESP32 firmware but I always receive this error while trying to connect to it using the flipper or using the serial console:

invalid header: 0xffffffff
invalid header: 0xffESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)

Will this board work?

justcallmekoko, 2025-04-16:
The Marauder firmware is not currently written to accommodate the C3.

closed 2025-04-16 by justcallmekoko

ATOMNFT, 2025-05-11:
I've been working on getting this board working, and have managed to get to work with the flipper zero companion app. I've added the C3 in The sketch as a new board option. The board I'm using is the ESP32-C3 super mini with rgb led and antenna interface from AliExpress. I will be testing to make sure sd card module will work with it as well. I will be adding this to my repo soon.