justcallmekoko/ESP32Marauder #712: compiling error 'esp_event_send_internal' was not declared in this scope

doyavivo, 2025-04-15:
I try to compiling ESP32 marauder using arduino 1.8.19 but error compiling message:

exit status 1
'esp_event_send_internal' was not declared in this scope

Any suggestion should I do to fix this issue?

justcallmekoko, 2025-04-16:
What version of the arduino esp32 core are you using? What board are you trying to compile for? What esp32 module is this for? Have you modified your arduino esp32 boards file as shown in the documentation?

Pro-Control, 2025-05-07:
Same for me I flow instructions 

https://github.com/justcallmekoko/ESP32Marauder/wiki/installing-firmware-from-source

Pro-Control, 2025-05-07:
update 

After going back to the instructions 
and make sure of flowing, 

1-   is version 2.0.10
2-  With any text editor, open C:\Users\<USERNAME>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\platform.txt
Add -w to the following compiler settings
build.extra_flags.esp32
build.extra_flags.esp32s2
build.extra_flags.esp32s3
build.extra_flags.esp32c3
Add -zmuldefs to the following compiler settings
compiler.c.elf.libs.esp32
compiler.c.elf.libs.esp32s2
compiler.c.elf.libs.esp32s3
compiler.c.elf.libs.esp32c3

I get this error 

`In file included from C:\Users\alial\Documents\esp32_marauder\Display.h:12,
                 from C:\Users\alial\Documents\esp32_marauder\Display.cpp:1:
c:\Users\alial\Documents\Arduino\libraries\JPEGDecoder\src/JPEGDecoder.h:48:20: error: conflicting declaration 'fs::SPIFFSFS LittleFS'
     #define SPIFFS LittleFS
                    ^~~~~~~~
C:\Users\alial\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\SPIFFS\src/SPIFFS.h:39:21: note: in expansion of macro 'SPIFFS'
 extern fs::SPIFFSFS SPIFFS;
                     ^~~~~~
In file included from c:\Users\alial\Documents\Arduino\libraries\JPEGDecoder\src/JPEGDecoder.h:47,
                 from C:\Users\alial\Documents\esp32_marauder\Display.h:12,
                 from C:\Users\alial\Documents\esp32_marauder\Display.cpp:1:
C:\Users\alial\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\LittleFS\src/LittleFS.h:39:23: note: previous declaration as 'fs::LittleFSFS LittleFS'
 extern fs::LittleFSFS LittleFS;
                       ^~~~~~~~
C:\Users\alial\Documents\esp32_marauder\WiFiScan.cpp: In member function 'NimBLEAdvertisementData WiFiScan::GetUniversalAdvertisementData(WiFiScan::EBLEPayloadType)':
C:\Users\alial\Documents\esp32_marauder\WiFiScan.cpp:62:71: error: no matching function for call to 'NimBLEAdvertisementData::addData(std::__cxx11::string)'
         AdvData.addData(std::string((char *)AdvData_Raw, 7 + name_len));
                                                                       ^
In file included from c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertising.h:38,
                 from c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:305,
                 from C:\Users\alial\Documents\esp32_marauder\WiFiScan.h:14,
                 from C:\Users\alial\Documents\esp32_marauder\WiFiScan.cpp:1:
c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertisementData.h:38:10: note: candidate: 'bool NimBLEAdvertisementData::addData(const uint8_t*, size_t)'
     bool addData(const uint8_t* data, size_t length);
          ^~~~~~~
c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertisementData.h:38:10: note:   candidate expects 2 arguments, 1 provided
c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertisementData.h:39:10: note: candidate: 'bool NimBLEAdvertisementData::addData(const std::vector<unsigned char>&)'
     bool addData(const std::vector<uint8_t>& data);
          ^~~~~~~
c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertisementData.h:39:10: note:   no known conversion for argument 1 from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to 'const std::vector<unsigned char>&'
C:\Users\alial\Documents\esp32_marauder\WiFiScan.cpp:84:61: error: no matching function for call to 'NimBLEAdvertisementData::addData(std::__cxx11::string)'
         AdvData.addData(std::string((char *)AdvData_Raw, 17));
                                                             ^
In file included from c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertising.h:38,
                 from c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:305,
                 from C:\Users\alial\Documents\esp32_marauder\WiFiScan.h:14,
                 from C:\Users\alial\Documents\esp32_marauder\WiFiScan.cpp:1:
c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertisementData.h:38:10: note: candidate: 'bool NimBLEAdvertisementData::addData(const uint8_t*, size_t)'
     bool addData(const uint8_t* data, size_t length);
          ^~~~~~~
c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertisementData.h:38:10: note:   candidate expects 2 arguments, 1 provided
c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertisementData.h:39:10: note: candidate: 'bool NimBLEAdvertisementData::addData(const std::vector<unsigned char>&)'
     bool addData(const std::vector<uint8_t>& data);
          ^~~~~~~
c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertisementData.h:39:10: note:   no known conversion for argument 1 from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to 'const std::vector<unsigned char>&'
C:\Users\alial\Documents\esp32_marauder\WiFiScan.cpp:109:61: error: no matching function for call to 'NimBLEAdvertisementData::addData(std::__cxx11::string)'
         AdvData.addData(std::string((char *)AdvData_Raw, 15));
                                                             ^
In file included from c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEAdvertising.h:38,
                 from c:\Users\alial\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:305,
[... 77 lines cut ...]
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
C:\Users\alial\Documents\esp32_marauder\WiFiScan.cpp:3258:17: error: 'class NimBLEScan' has no member named 'setAdvertisedDeviceCallbacks'
       pBLEScan->setAdvertisedDeviceCallbacks(new bluetoothScanAllCallback(), false);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "SD.h"
  Used: C:\Users\alial\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\SD
  Not used: C:\Users\alial\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: no matching function for call to 'NimBLEAdvertisementData::addData(std::__cxx11::string)'`