justcallmekoko/ESP32Marauder #716: I have just a regular ESP development board but I flashed on Marauder I use it with no screen in a terminal but it seems super basic is there any command libraries you have that you might be able to pass along or maybe give me some advice or something I'm not doing correctly

Bozofromnom, 2025-04-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Aboalfadl, 2025-04-16:
Hi, I think i have the same regular ESP dev board, its ESP32-C3 SuperMini board, i don't know if the Marauder firmware will work on it or not because I tried to flash it but it doesn't work.
What type of board and version do you have ?

Bozofromnom, 2025-04-16:
> I have no idea what you're talking about

Just a bare esp32e dev board

Su-Qingmu, 2025-04-27:
我也是这种情况，我使用的ESP32-C3-WROOM-02U，但是我希望在使用时把它藏起来，用手机进行控制，因此我不需要屏幕以及其他外设，有没有适用这种情况下的bin固件版本

I'm in the same situation. I'm using the ESP32-C3-WROOM-02U, but I want to hide it during use and control it via my phone, so I don't need a screen or other peripherals. Is there a suitable .bin firmware version for this scenario?

justcallmekoko, 2025-04-27:
@Bozofromnom 

If you asking about using commands over serial to control the Marauder firmware, we have the entire CLI manual documented in the Wiki along with several workflows for different analysis operations.

closed 2025-05-30 by justcallmekoko