justcallmekoko/ESP32Marauder #726: Just getting the backlight on on tft lcd 2.8 inch using elecrow hmi 2.8inch

Trinnss, 2025-04-22:
The Serial / Commands are working but only getting this given screen.[image]

justcallmekoko, 2025-04-27:
From where did you receive this device and what firmware is running on it. Since I don't recognize the hardware, I wouldn't know if this is running some sort of unofficial port of the firmware. 

To me this looks like it is probably some sort of SPI conflict which is likely the result of a bin file being installed on a hardware is wasn't meant to run with.

Trinnss, 2025-04-28:
So I compiled the repository of yours project with all the libraries(given there) installed. I have tried multiple times by the watching the wiki but everytime it is failing. Can U make a firmware(bin) file for this display so that there is no conflict.