justcallmekoko/ESP32Marauder #734: Ap clone spam keeps crashing on version 1.4.5

Matizek1, 2025-04-27:
I changed to version 1.4.5 and it's still crashing.As i said i use 2.8 CYD

justcallmekoko, 2025-04-27:
Can you show a video from start to finish how you prepare you clone spam and execute it? I need to see how you're acquiring the data to clone

Matizek1, 2025-04-27:
> Can you show a video from start to finish how you prepare you clone spam and execute it? I need to see how you're acquiring the data to clone

Could i type what i did?

Matizek1, 2025-04-27:
> Can you show a video from start to finish how you prepare you clone spam and execute it? I need to see how you're acquiring the data to clone

It seems that i can't get the video to upload

Matizek1, 2025-04-28:
> Can you show a video from start to finish how you prepare you clone spam and execute it? I need to see how you're acquiring the data to clone

so maybe i will type it here beacuse i can't get the video to upload:
first i load the AP's from my sd card then go to wifi,general,select ap's and select multiple AP's then i go back and go to attacks and choose AP clone spam and after a second i get a crash.
I hope that helped you in understanding my issue if i need to explain something more please let me know.
Regards

justcallmekoko, 2025-04-28:
What happens if you don't load the APs from a file but instead scan them and used one of the scanned APs?

Matizek1, 2025-04-29:
> What happens if you don't load the APs from a file but instead scan them and used one of the scanned APs?

It works normally

Matizek1, 2025-05-02:
> What happens if you don't load the APs from a file but instead scan them and used one of the scanned APs?

So yeah when will you fix it?

closed 2025-05-12 by Matizek1